JMRI/JMRI #12459: add option(s) when restoring block values

mstevetodd, 2023-09-28:
For background, see users group conversation here: https://groups.io/g/jmriusers/topic/101507803

JMRI saves block values to "blockvalues.xml" at shutdown (jmri.jmrit.display.layoutEditor.BlockValueFile.writeBlockValues()), then tries to restore those values at startup (jmri.jmrit.display.layoutEditor.BlockValueFile.readBlockValues()).
readBlockValues() currently includes this logic:
`check if all powermanagers are turned on, if they are, we should expect blocks with values to be occupied`
and it skips all updates if power is on and all saved blocks are not occupied. There are times when it would be good to restore the values regardless of occupancy or power state.

I suggest adding a mechanism to allow the user to override this requirement.

In the linked thread, it was suggested we add it as a Preference, or add it in the Blocks Table menu.

mstevetodd, 2023-09-28:
The Blocks Table menu currently looks like this:
[image]
If I understand the suggestion, we'd add a new menu option "Values" which opened a dialog with options such as "Restore Always", "Restore Occupied Only", "Restore Only if All Occupied". (wording is WIP, to be improved based on actual restore solution)

danielb987, 2023-09-29:
Looks good to me.

dsand47, 2023-09-29:
It does not need to be a dialog.  It can be a list of menu items.  The Logix table has an options menu with two sets of items.
[image]

I recommend that the choice be stored in the user-interface.xml file rather than the layout configuration xml file.  This eliminates any backward compatible file issues.  The code for this approach is available in /jmri/jmrit/beantable/LogixTableAction.java

danielb987, 2023-10-07:
Is this issue completed?

mstevetodd, 2023-10-07:
Yes

closed 2023-10-07 by danielb987