JMRI/JMRI #12465: Resizing Digital Fast Clocks with Pause Button Obscures Pause Button

DeanCording, 2023-09-30:
When you create a fast clock with an LCD, Nixie, or Pragotron fast clock display with a Pause button, resizing the clock window in certain directions results in the time display expanding to fill the entire window with the Pause button pushed out of the window frame.

See attached screen recording demonstrating the problem.

[Screencast_20230930_152937-1.webm](https://github.com/JMRI/JMRI/assets/251170/b65e0ce5-f20e-4868-bb5e-8918252a7d14)

bobjacobsen, 2023-11-02:
PR #12577 has the proposed fix.

bobjacobsen, 2023-11-06:
I think this is now fixed.  @DeanCording could you try a [development release](https://builds.jmri.org/jenkins/job/development/job/packages/) and let me know please?

DeanCording, 2023-11-08:
Looks good.

DeanCording, 2023-11-08:
Looks good.

DeanCording, 2023-11-08:
Looks good.

closed 2023-11-08 by DeanCording