JMRI/JMRI #12486: Version 2 Digitrax Series 7 Extended

SPJMaker, 2023-10-06:
Clean up and revise
  Cleanup 9/29/23:
       soundOptions7
       soundSettings7
       PaneDigitraxSoundExtSeries7
	add Plain SDH167 decoder
					
[Digitrax_0SFX_Series7.zip](https://github.com/JMRI/JMRI/files/12834701/Digitrax_0SFX_Series7.zip)

bobjacobsen, 2023-10-06:
@SPJMaker Does this also require a .xsl file?

SPJMaker, 2023-10-06:
No, it does not.
It is just the clean up.

The XSL you drafted would not validate or install.

John McMasters
AnPRR
***@***.***

bobjacobsen, 2023-10-06:
This is missing the "xml/decoders/digitrax/functionMapSettings7.xml" file. Could you attach that one please?

SPJMaker, 2023-10-07:
Sorry about that!  I thought I removed the functionmap settings.
Here is the zip with only the cleanup.

John McMasters
AnPRR
***@***.***

danielb987, 2023-10-11:
@SPJMaker 
Unfortunately you cannot attach a file to an email. GitHub doesn't accept that. You have to log in to GitHub and attach the file to a reply in this issue with your web browser.

SPJMaker, 2023-10-11:
[Digitrax_0SFX_Series7.zip](https://github.com/JMRI/JMRI/files/12868793/Digitrax_0SFX_Series7.zip)

SPJMaker, 2023-10-11:
ok here us the revision

bobjacobsen, 2023-10-11:
See PR #12513

closed 2023-10-12 by danielb987