JMRI/JMRI #12514: Create decoder Fabricant LEB

Speedjf37, 2023-10-12:
Hi !
How to Create decoder Fabricant LEB ?
http://lebmodelisme.com

With number for fabricant  and acces to add files for decoders ?

Thanks

JFP

bobjacobsen, 2023-10-12:
@Speedjf37 (My French is weak. My apologies if I've misunderstood the LEB web page)

These look like signal decoders.  Do they have CVs that are used to configure them?  If so
  - Please provide a URL for a manual that shows the CVs
  - Please read the value of CV7 and CV8 from them.
With that info, I think I can put together a basic programming definition for them in English. If you'd like it in French, I can show you where to put the translations.

As signal decoders, do you want to use them for SNCF signals?  JMRI's definitions for those are described here:
https://www.jmri.org/xml/signals/SNCF-2015/index.shtml
and here
https://www.jmri.org/xml/signals/SNCF-2015/aspects.xml
if that's what you're looking for, let me know and I can provide more info on how to set them up.  Or are you looking for something else?

Speedjf37, 2023-10-13:
I want to insert in JMRI decoders the files join

[leb_JMRI.zip](https://github.com/JMRI/JMRI/files/12895342/leb_JMRI.zip)

bobjacobsen, 2023-10-13:
Thank you for this!

There are a couple issues that need to be resolved before I can merge these:

 - The `Leb_DECSIGBF1_OP1.xml` file references a `leb/TableAllumageOptions.xml`file, but I didn't get a file by that name.  Could you provide it please? Or is should this be referencing one of the existing files?
 - The `xml/decoders/Leb_DECSIGBF1_V4.xml` file has a duplicate definition of Table Allumage Personnalisee".  Should the second one be "2ID separes"?  or something else?
 ```
			<variable item="Table Allumage Personnalisee" CV="3" mask="XXVXXXXX" default="0">
				<xi:include href="http://jmri.org/xml/decoders/parts/enum-disabledEnabled.xml"/>
				<label>Table Allumage Personnalisee</label>
			</variable>
			<variable item="Table Allumage Personnalisee" CV="3" mask="XVXXXXXX" default="0">
				<xi:include href="http://jmri.org/xml/decoders/parts/enum-disabledEnabled.xml"/>
				<label>2ID separes</label>
			</variable>
 ```
 
 There were also a couple of small validation changes that I fixed.  You can seem them in [this commit](https://github.com/JMRI/JMRI/commit/1cf9b829b21f8d8a0703e4955381475d948eaf55).
 
 Thanks again for the contribution.  If you'd like to be included in the [JMRI Acknowledgement page](https://www.jmri.org/help/en/Acknowledgements.shtml), please send me your name.

bobjacobsen, 2023-10-13:
Also, if you have one of these, could you read the value of CV8 please?  That will help me set up automatic identification of these decoders.

Speedjf37, 2023-10-13:
Thanks for return !
Tomorrow i send files modifie and missing
CV8 is 13 for DIY by default

my name : PLISSON Jean-François

Speedjf37, 2023-10-14:
Hi,

How to create number for fabricant LEB ?
Join modified and missing files

[leb_JMRI.zip](https://github.com/JMRI/JMRI/files/12904251/leb_JMRI.zip)

JF

bobjacobsen, 2023-10-14:
Thank you!  There are still two problems.

First, in the file xml/decoders/Leb_DECSIGBF1_V4.xml there are still duplicate definitions:
```
			<variable item="Table Allumage Personnalisee" CV="3" mask="XXVXXXXX" default="0">
				<xi:include href="http://jmri.org/xml/decoders/parts/enum-disabledEnabled.xml"/>
				<label>Table Allumage Personnalisee</label>
			</variable>
			<variable item="Table Allumage Personnalisee" CV="3" mask="XVXXXXXX" default="0">
				<xi:include href="http://jmri.org/xml/decoders/parts/enum-disabledEnabled.xml"/>
				<label>2ID separes</label>
			</variable>
````

Note that the 2nd one has `item="Table Allumage Personnalisee"` but has a different label: `<label>2ID separes</label>` I think the solution is to change the item attribute to `item="2ID separes"`. Let me know if you want me to do something else.

The second problem is in the xml/decoders/Leb_DECSIGBF1_OP1.xml file:
```
			<variable CV="1" mask="XXVVVVVV" comment="Decoder A address " item="Short A Address " default="03">
<!--			<variable item="Low 7 bytes of Address CV1" CV="1" mask="XVVVVVVV" comment="Decoder address" default="01"> -->
				<decVal min="1" max="63"/>
				<splitVal highCV="9"/>
				<label>Decoder A Address</label>
				<comment>Decoder A address </comment>
			</variable>
```
You can't have both `<decVal>` and `<splitVal>` elements in a variable.  You can only have one or the other.  I'm going to remove the <splitVal>`. If that's not consistent with what you're trying to do here, could you let me know and I'll make the change.

Thanks again for this.  I'm going to start the process to merge this into JMRI so it will be in the next release in early November.

Speedjf37, 2023-10-14:
2 errors make change .
Update somme files.
How to limit value ?
 CV 1  >=3 <= 63
 CV 9  >=0 <= 7

JF

Speedjf37, 2023-10-14:
[leb_JMRI.zip](https://github.com/JMRI/JMRI/files/12907015/leb_JMRI.zip)

bobjacobsen, 2023-10-15:
These changes are in #12523 now.

In the future, if you need to make any more changes, could you please start with the files attached here?  I had to make some changes to get them to validate.  If you use these, I won't have to make those changes again.  Thank you.

[Archive.zip](https://github.com/JMRI/JMRI/files/12910135/Archive.zip)

bobjacobsen, 2023-10-15:
You can add "min" and "max" attributes to limit the values, but they're not very effective.  People can still enter values that are too big or too small into the text fields.  DecoderPro will set the value to the min or max, but people don't always notice that.

```
<decVal min="3" max="63"/>
```

Speedjf37, 2023-10-16:
Hi

Ok for new change with files in Archive.zip.

Thanks
JF

danielb987, 2023-10-16:
> You can add "min" and "max" attributes to limit the values, but they're not very effective. People can still enter values that are too big or too small into the text fields. DecoderPro will set the value to the min or max, but people don't always notice that.
> 
> ```
> <decVal min="3" max="63"/>
> ```

This seems like something that should be fixed. That if the user enters a wrong value, a message box should be shown to alert the user.

closed 2023-10-19 by bobjacobsen