JMRI/JMRI #12529: LCC Firmware Downloader window

KenC57, 2023-10-17:
While the help page talks about an 'Open' button for finding the node to update, the tool uses a select list. However it doesn't correctly show in the window unless dragged quite wide. Here is the default open of the tool.
[image]
It doesn't look like a select, it looks like another line on the screen. Dragging it wide enough turns to:
[image]
The empty box just doesn't show up very clear.
This is on Win11 with Preferences->Display->GUI->Metal
as the selection. I could see many users being confused by the view and not know the trick of resizing things to find hidden features.

crosland, 2023-10-17:
I haven't seen this exact issue, but I have seen the selector change width to be too narrow to display the node details. I haven't tried this in the latest JMRI test release.

atanisoft, 2023-11-10:
you might consider cross-posting this to https://github.com/openlcb/OpenLCB_Java which provides the LCC functionality.

As for the wide selector box, it is due to assigning user names (64 characters) to the nodes which when combined with the node ID it grows considerably. Assigning names to the nodes is a good thing as it can help identify them easily.

Perhaps another option would be to add "firmware downloader" as an option in the node list on right click if it isn't there already.

bobjacobsen, 2023-11-12:
There's an improvement to the layout in PR #12605

bobjacobsen, 2023-12-05:
@Kenc57 Did PR #12605 fix this?  I.e. can we close it?

KenC57, 2023-12-21:
Yes it has now been fixed. Tested with 5.7.1+

closed 2023-12-21 by KenC57