JMRI/JMRI #12535: PanelPro: UI issues with Send Frame window

RobertPHeller, 2023-10-19:
The Send Frame window (opened under LCC menu of PanelPro) is too wide. The section titled "Send OpenLCB event message:" is too wide.  It really should be in (at least) two rows.  At least on my monitors, the window wants to be wider than my screen.    Also, the text entry for "Send Event Produced" wants to shrink  to zero width when the existing text is deleted -- this is probably wrong (something is illset for its geometry management) -- I almost always need to put something else there.  (Also: *feature request*: would it make sense to include a *Paste* button to match the *Copy* buttons on (for example) the CDI editor?)

bobjacobsen, 2023-11-01:
Could you try this head of master in GitHub or with a recent [development release](https://builds.jmri.org/jenkins/job/development/job/packages/) to see if it's still happening?  If so, could you attach some screen shots.  Thanks!

bobjacobsen, 2023-12-18:
Fixed in 5.5.8

closed 2023-12-18 by bobjacobsen