JMRI/JMRI #12547: FastClock-Telegram missing

Kruemelbahn, 2023-10-23:
I'm just figuring out the possibilities of FastClock.
Everything works fine and FastClock-telegrams are send properly, 
except: when the time changes from minute 59 to minute 00 no telegram is send.
Next telegram I receive is for minute 01.
I've checked it with several hours, everything the same.

Kruemelbahn, 2023-10-23:
The hardware I use is a FastClock with LED display that I built myself using the Arduin lib for Loconet (current version).
JMRI runs on my PC and the absence of the telegram could be observed in the JMRI Loconet monitor. 
I've attched the FastClock-Seetings as shown in the picture. Selected is "Internal Timebase" and "synchronize clock with Loconet"
[image]

icklesteve, 2023-10-23:
I can reproduce this with a Simulated LocoNet Connection, eg.

```
Request Fast Clock is Synchronized, Running, rate is 4:1. Day 22, 17:57. Last set by ID 0x00 0x00 (0).
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available.
Request Fast Clock is Synchronized, Running, rate is 4:1. Day 22, 17:58. Last set by ID 0x00 0x00 (0).
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available.
Request Fast Clock is Synchronized, Running, rate is 4:1. Day 22, 17:59. Last set by ID 0x00 0x00 (0).
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available.
Request Fast Clock is Synchronized, Running, rate is 4:1. Day 22, 18:01. Last set by ID 0x00 0x00 (0).
	Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available.
```

icklesteve, 2023-10-23:
In LnClockControl at https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/loconet/LnClockControl.java#L242 there's an interesting comment,

```
// skip the correction if minutes is 0 because Logic Rail Clock displays incorrectly
// if a correction is sent at zero minutes.
```

which explains why the JMRI Memory value updates but no outward message is sent.

Looking at the LocoNet PE Spec, https://www.digitrax.com/static/apps/cms/media/documents/loconet/loconetpersonaledition.pdf

`... all attached display devices keep a current clock calculation based on this SYNC read value, i.e. devices MUST not continuously poll the clock SLOT to generate time, but use this merely to restore SYNC and follow current RATE etc. This clock slot is typically "pinged" or read SYNC'd every 70 to 100 seconds ...`

so ( unfortunately for your implementation ) it's not actually a requirement for the rate to be sent every fast minute change.

This subject may well have cropped up before, so we'll see if anyone else has any thoughts on this.

Steve.

devel-bobm, 2023-10-24:
I don't remember any time when this has popped up, but this code change goes back before the move to Git.  With code this old, I do not have an easy way to know who made the changes and when they made the changes.  And I certainly do not know why the change occurred!  (And I hope I was not involved in this "fix"...)

FWIW, this "hardware-specific" modification is, in my opinion, problematic.  Why does the LNFC _force_ the behavior on _all_ LocoNet clocks, including the hardware that the O.P. is working with?  Probably this code change came about because the code change _could_ be done without a "connection preference".  That sort of concern has been a problem in the past.  Now, with a specific problem with a specific hardware platforn, that choice doesn't seem a great one.

One "solution" would be to implement the hardware to be a "LocoNet Personal Edition 1.0 Specification"-compliant fast-clock.  The hardware would need to "predict" the fast-minutes-change, and act even though the "fast-clock master" might not send the fast-minute "tick".  Ideally, this would "solve the problem" without needing any JMRI change.

An alternate "solution" would be to make an "Additional connection setting" in the LocoNet connection, such that the user can "enable" or "disable" the "LNFC feature-avoidance".  When enabled (the default setting!), the JMRI code would behave the way it has for so many years.  When disabled, the JMRI code would "skip" the code that checks the number of minutes, so that all 60 minutes generate the LocoNet messages (instead of 59 of the 60 minutes).   

(And I would expect that (at least some of the versions of) the LNFC hardware is "not compatible" with the new hardware.  That would need to be CLEARLY documnted!)

bobjacobsen, 2023-10-24:
I think @kiwi64ajs Alex Shepherd worked in this area originally.

kiwi64ajs, 2023-10-24:
Yes, I did quite a bit in the early days to decode and debug the LocoNet FastClock message field values, and the odd behaviour that makes the DCS100 hardware tick too fast compared to a DTX00 throttle.
So, are you commenting on the FastClock implementation in JMRI or in the Arduino LocoNet library? The regular sending of the FastClock messages is probably related to the FactClock time correction mode in which the DCS100 ticking too fast is corrected every fast minute roll-over.
However, based on the code comment in JMRI, there seems to have been an issue with the LogicRail product behaviour that is somehow avoided by suppressing the hourly 0-minute roll-over. I do recall there being an issue with the LogicRail product not displaying the time field values the same as a DTX00 throttle, which I think was an off-by-one error, so perhaps that explains the observed behaviour with JMRI as someone "fixed" it or at least added that work-around.

KenC57, 2023-10-24:
All I can recall is back in the 2004-2008 timeframe, there was a fix in JMRI
to make up for a bug in the Digitrax command station fast clock. That's how
I recall it. And it had to do with the rollover. That's about all I remember
there was a something. I was working on the NCE fast clock parts and had to
be careful to avoid following all of what the LocoNet clock did as an
example.

Kruemelbahn, 2023-10-25:
State of current situation on my side:
I got my FastClock now working with JMRI after some corrections in my local LocoNET library (taken a copy from V1.1.13).
Two problems were fixed in my local library :
- FastClock telegrams call 'doNotify(1)' in "processMessage()" only if in byte 10 (<CLK_CNTRL>) bit 6 is set - this is missing in the JMRI telegrams, they always send 0 instead of 40h.
- because the telegram from JMRI is missing when changing from minute 59 to minute 00, my clock now continues to run internally and synchronizes with the next telegram from JMRI (on my side I'm waiting for telegram and not polling them) 
  But for this to work properly, "process66msActions()" had to be modified: JMRI telegrams start in <FRAC_MINSH> with the value 3 instead of the default 7Fh.
  Due to the small start value, the counter reaches only a maximum value of 10 in one minute, while waiting for the next telegramm a maximum of 17 was reached.
  The condition 'if (fcSlotData.frac_minsh & 0x80)' is never met, 'doNotify(0)' is not called.

Currently my FastClock is running now, IMHO either adjustments in JMRI or the LocoNET library are useful now.

Kruemelbahn, 2023-12-12:
Is there any consideration of correcting either the JMRI library or the LocoNet library accordingly?

kiwi64ajs, 2023-12-12:
Several things:

1) Rereading this issue, I don’t see any mention of which LocoNet DCC Command Station is being used. I’m pretty certain that for a LocoNet Fast Clock, there is an assumption that there is a Command Station present to act as the FastClock master node.

2) Depending on the desired behaviour and mix of LocoNet devices used to display the Fast Time (including JMRI and DTx00 throttles etc) you may or may not want to correct the FastClock Time each Fast-Minute roll-over.

If you are using wireless throttles via IR or Simplex Radio then the FastClock in the throttles will get out of sync with the Command Station. If all devices are on LocoNet, then it's not such an issue, but it still can be problematic.

3) As per the LNPE Spec, you should NOT be reliant on receiving LocoNet Fast Clock Slot updates on every Fast Minute rollover to drive your FastClock slave device. it should handle this itself.

Can you please describe your LocoNet network devices and the DCS Command Station model and email me your LocoNet library file changes so I can diff them and maybe try to replicate your observations?

It's been a very long time since I touched the FastClock code, let alone tried to debug it with JMRI… ;)

HTH

Alex Shepherd

Kruemelbahn, 2023-12-13:
Thanks for your reply, Alex.

My hardware situation is as follows:
- Raspberry Pi 3B+, running as accesspoint (software by Steve M.Todd), running JMRI V5.5.5
- using a locobuffer (ArduinoNano with software LocoLinx from original LocoNet-library, current version)
- connected via LocoNet to TwinCenter by Fleischmann (similar to Intellibox 1 by Uhlenbrock) as DCC-Commandstation
[image]
Since this DCC-Commandstation is unable to create and send the FastClock-telegrams, I activated this in JMRI.

I build-up a FasctClock-Slave (named by me as "NebenUhr") which shall - among other expermental possiblities - receive a FastClock-telegram to show the time (see: https://github.com/Kruemelbahn/Nebenuhr).

1.) During setup I found a problem: FastClock-Notification doesnot work.
I monitored the FastClock-telegram from JMRI, which was send to LocoNet and it looks well.
But I didnot receive the expected notification "notifyFastClock" from the LocoNet-Library (I'm using the current version).
Taking a deeper look into the libraray, I found out, that calling "notifyFastClock" needs a valid Fastclock-telegram:
According to the LNPE-Spezification JMRI should send the FastClock-telegram with Bit 6 set in <CLK_CNTRL>, but <CLK_CNTRL> is always 0
And the library needs bit 6 set:

```
void LocoNetFastClockClass::processMessage(lnMsg* LnPacket)
{
	if ((LnPacket->fc.slot == FC_SLOT) && ((LnPacket->fc.command == OPC_WR_SL_DATA) || (LnPacket->fc.command == OPC_SL_RD_DATA)))
	{
		if (LnPacket->fc.clk_cntrl & 0x40)
...

```
So I changed the function processMessage to react independly from <CLK_CNTRL>.
This works and so I received the expected notification.

2.) And this leeds to the next problem: minute zero was not send.
You gave the hint, that I "... should NOT be reliant on receiving LocoNet Fast Clock Slot updates on every Fast Minute rollover to drive your FastClock slave device".
So I took a closer look to "process66msActions", which should do the work to continue counting my time without receiveing a FastClock-telegram.
"process66msActions" should also call "notifyFastClock" but doesnot within 60 (FastClock-)Seconds.
I found out, that following condition was never reached:

```
void LocoNetFastClockClass::process66msActions(void)
{
	// If we are all initialised and ready then increment accumulators
	if (fcState == FC_ST_READY)
	{
...
			if (fcSlotData.frac_minsh & 0x80)
...

```

After analysing the JMRI-FastClock-Telegram and making some analysis inside the library-code,
I noticed that JMRI starts with 'fcSlotData.frac_minsh = 3' instead of FC_FRAC_RESET_HIGH (0x78),
Therefore fcSlotData.frac_minsh is always less than 0x80 and fcSlotData.frac_minsh counts in maximum up to 10 within one (FastClock-)Minute. 
So the 'fcSlotData.frac_minsh' will be increased by maximum of 6, independant from fcSlotdata.clk_rate
Therefor I changed the function process66msActions to enable the call of notifyFastClock.

This works and so my FastClock-Slave works as excpected: if a telegramm is missing, FastClock-time will continue counting.

My handling with the LocoNet-library:
A.)  First I have made a copy of LocoNetFastClockClass, named it LocoNetFastClockClassKS, implemented my changes an put it in a new LocoNet-Library, which can be found here: https://github.com/Kruemelbahn/LocoNetKS.
B.) I also forked the original LocoNet-Library, made the corrections, add comments and asked for a pull-request (https://github.com/mrrwa/LocoNet/pull/40). My corrections should be compatible with the excisting library, because I added a new function "initJMRI()". Without calling this, the library should work as before.

Expectation for JMRI:
On the other hand, the changes are due to shortcomings in JMRI. At least position 1 could simply be corrected within JMRI.

HTH
Michael

Pugwash1, 2023-12-18:
@Michael-Richardson 

There are :
"EF 0E 7B" (Write fast clock slot)  messages which are issued by a client to correct a server.
"E7 0E 7B" (Response to time request) messages which are issued by a server in response to a request 
for the time ("BB 7B 00 3F")

The server only listens for clock updates and clock requests and the client only listens for responses.

The Arduino Library is correct. 

Remember, if the code is to be useful it must be fully compatibly to the prottype, else it is only useful to a section of the community.

It is generally accepted that its impolite to clog the loconet with messages evey fast clock minute, just as your pc doent ask a NTP server for the time every hour or even every day.
You listen for a fast clock response, and set your clock. You update it internally and if you dont see another message within (n seconds - where n is around 90 for genuine digitrax throttles) then you request the time from the server.

The JMRI Loconet fast clock can be sourced from the Digitrax CS or Internallly.
If it is Internal it can reset the DS fast clock with its own time (either of the check boxes do the same thing).

If you want to upgrade the JMRI fast clock to act as a loconet server that can be done, by sending a response with out recieving a request every 70-80 seconds and answering a request about 2 seconds after receiving a request. The delay is required as you cannot stop the CS from responding. so you respond after and watch the analog clocks spin...

"

closed 2023-12-21 by Kruemelbahn