JMRI/JMRI #12561: Seeing differences in the web panel vs the JMRI panel

Cliff-Anderson, 2023-10-27:
In some cases, the web panel rendering of a Layout Editor panel does not render the background image that is visible on the JMRI computer's screen.  

At the suggestion of @mstevetodd in the message https://groups.io/g/jmriusers/message/222302 an example panel which can be used to replicate the issue is attached as a zip of a specially constructed Config Profile folder.  

Steve's requested screen shots are inside the zip file.  

A copy of the log file is also included.  

I hope this helps.

[web_panel_vs_JMRI_panel.jmri.zip](https://github.com/JMRI/JMRI/files/13193852/web_panel_vs_JMRI_panel.jmri.zip)
[session.log](https://github.com/JMRI/JMRI/files/13193857/session.log)

mstevetodd, 2023-10-28:
Thanks, @Cliff-Anderson 

A quick glance shows a likely culprit, you have special characters in your file name. In particular, the "&" is going to create issues with retrieving that file over the web.
Can you try it again without the "&"?

mstevetodd, 2023-10-28:
@Cliff-Anderson 
Actually, the single quote is the initial issue. Here's the error from the browser console:
[image]

The workaround is to remove the single quote and the ampersand from the background filename.
Since it looks like you're generating this from AnyRail, I suggest removing both from the AnyRail filename and re-exporting.

I'll still take a look at handling those better, but it might be a while.

I find that special characters in filenames are seldom worth the headaches, so I avoid them without even thinking about it.

mstevetodd, 2023-10-28:
I removed only the single quote from the .jpg filename and from the icon URL in the .xml file, and it works fine now. See browser screenshot.
[image]

mstevetodd, 2023-10-29:
PR #12565 created to handle the apostrophe in url filenames

closed 2023-10-30 by danielb987

mstevetodd, 2023-10-31:
@Cliff-Anderson This fix has been merged and is available in the Development version now, and will be in v5.5.6 and later.

commit 59159670 (2023-10-30): Merge pull request #12565 from mstevetodd/master

handle apostrophe in url filenames, closes #12561