JMRI/JMRI #12564: Update Decoder Definitions (Menu Actions)

uli-2016, 2023-10-28:
When the update of the decoder definitions is started, the entire locomotive park is processed and all decoder definitions are checked  for replaceability. However, there should be an option to select only one locomotive park entry and to perform the update only for this one.

uli-2016, 2023-11-01:
Does anyone know a solution to this ?

bobjacobsen, 2023-11-01:
The current "Update Decoder Definitions" DecoderPro Debug item is handled by the `jmri.jmrit.roster.UpdateDecoderDefinitionAction` class. That could be modified to handle just one.

How would you like to invoke it?  It's a bit tricky because it changes data structures for an individual roster entry.  it shouldn't be done while that roster entry is open, so it shouldn't be done from the menus or panes of the individual roster entry.  Maybe from the pop-up (ctrl-press) window on the main roster?

uli-2016, 2023-11-05:
If possible: Select the roster ID and than po-up menu right mouse button, as like delete a roster