JMRI/JMRI #12569: Unable to Delete old LogixNG Module

slolitzky, 2023-10-30:
I attempted to delete an old LogixNG module by selecting Delete.  I expected a dialog box to appear and confirm the delete.  It didn't happen.  It will allow me to delete an LogixNG routine though.  I also notice that until I set the option to back to Select, I can't edit/focus on any of the other LogixNG modules.  

Based on input from one of the developers I tried to make sure there were no longer references to it in any of the other LogixNG items, but this didn't work.  I do get a Java Exception in the System Console.

**I am currently using one of the test builds 5.5.5plus+jenkings+20231028T0027Z.  It also appears on an older test build that I was using from 5.5.3.**

**### I think there should be a warning message at least and possibly some way to find what other references are calling the Module (if possible).**  

Here is the response I received on the message board from the developer.

_If you look at the system console you will see that the request to delete the module has been vetoed since there are still references to the module.  The references can be from other modules, LogixNGs or the LogixNG clipboard.  For example:  logixng.tools.swing.DeleteBean WARN  - In use by IQDA:AUTO:0006 - Call module : Do Not Delete
The veto process definitely has some issues.  Instead of displaying a message, it creates a Java exception.  This also leaves the Select menu on the Delete item.  Another issue is that you have to do a store/stop/start/load sequence between each delete to clean up the in-memory references.
Dave Sand__

[image]

danielb987, 2023-10-31:
Thank you for reporting this! It was a bug and it's fixed by PR #12571. PR #12571 is mostly about other stuff, but the fix for this issue was so simple so I included it in that PR.

closed 2023-11-01 by danielb987

danielb987, 2023-11-01:
PR  #12571 is now merged and the fix will be included in JMRI test release 5.5.6 and JMRI production release 5.6.