JMRI/JMRI #12572: loksound 5 decoder file AUX phase output

dvescovi1, 2023-10-31:
the ESU programmer for lokSound 5 (I tested with LokSound 5 micro KATO) has a setting for Function outputs called "Phase Shift" on AUX 2, 3 etc.
This is a slider and the "Phase" cv can be set from 0 to 63.
Decoder Pro for the same decoder on Function Outputs only a single check box called "Phase Reverse".
I do not know how the CV's map or if this is a problem for all loksound "5" decoders.

newHeiko, 2023-11-14:
Yes, that's a new feature from firmware 5.8.156 upwards (July 2023) and we haven't caught up with it yet.

For more info, see https://www.esu.eu/en/downloads/software/lokprogrammer/ (Click on "Additional information" for the 5.2.4 version)

I'm not saying I'll fix this - both because I'm notoriously bad at getting stuff done and because I'm not using those options at the moment, and also because I'm still figuring out firmware-version-dependent decoder definitions - but if no one else is taking care of it, I might have a go.

Heiko

bobjacobsen, 2025-03-03:
This was also mentioned in https://groups.io/g/jmriusers/message/240233