JMRI/JMRI #12603: Remote Browser shows UserName of Sensor (ToolTip?)  even when ToolTips are turned off

Paelgin, 2023-11-12:
Using JMRI 5.5.6 with Java 11.0.21 on Windows 10, a remote browser viewing a Webpage, of either a Layout Editor or Control Editor Panel, with a Sensor on it, will show the UserName when hovered over by the mouse pointer.  A Turnout will show the System Name on the remote browser when hovered over with the mouse.  

I assumed that this was ToolTip showing in the remote browser so I tried turning off ToolTips to no avail.
Phil 

[image]
[image]
[image]
[image]

mstevetodd, 2023-11-13:
@Paelgin 
What behavior are you asking for? It appears there is a "Show Tooltips for all items" at the panel level, then each item has its own "Show tooltip" option. What is supposed to happen for each combination of those? Which one takes precedence?

Paelgin, 2023-11-13:
For my remote browser view, I do not want to see any ToolTips or Sensor names (or Turnout System Names).  I don't know if it is actually ToolTips.  For example, when I made a custom ToolTip for one Sensor, the remote panel still showed the User Name, not the custom ToolTip.  

When editing the Panels, I unclicked the "Show Tooltip" on each Sensor by right-clicking and then I also set the Option to "Show icon tooltips" to "Never".  The remote browser still showed stuff that I do not want to show.

dsand47, 2023-11-13:
When the panel level tooltip option is changed, the change is also applied to all of the components.  

This also occurs for Layout Editor when switching between edit and display mode if "In Edit Mode only" or "Not in Edit Mode only" is selected.

Custom tooltips are stored in a `<tooltip>` element within the icon element.

mstevetodd, 2023-11-13:
> When the panel level tooltip option is changed, the change is also applied to all of the components.
>
> This also occurs for Layout Editor when switching between edit and display mode if "In Edit Mode [...]

Thanks, @dsand47 . So from that it sounds like the web page can ignore the panel-level flag and simply follow the flag for each element. Is that correct?

dsand47, 2023-11-13:
> Thanks, @dsand47 . So from that it sounds like the web page can ignore the panel-level flag and simply follow the flag for each element. Is that correct?

That is my understanding.

The tooltip process was re-designed at release 4.99.3.  

The current details are at https://www.jmri.org/help/en/package/jmri/jmrit/display/Tooltips.shtml

Paelgin, 2023-11-23:
I tried to upload a test file with two panels (changed from .xml to ,txt) that I made to show my tooltips question.  When loaded in JMRI (5.5.6 with Java 11.0,16, Digitrax Simulator) with WebServer running, I can (on a remote computer) log into the JMRI webServer and look/control those panels. When I hover the mouse over the Sensor icon, the ToolTip shows up even though I have Tooltips individually shut off.  That is the behavior I do not want.

Incidentally, the ToolTips for Text icons do not show up on the remote panel (and I'm glad).

[2023-11-12-1140-TestXML-ForRemoteToolTipIssue.txt](https://github.com/JMRI/JMRI/files/13452345/2023-11-12-1140-TestXML-ForRemoteToolTipIssue.txt)

Phil

closed 2024-02-12 by github-actions[bot]