JMRI/JMRI #12621: "Import Decoder File" falls apart when decoder file includes other xml files

newHeiko, 2023-11-14:
Don't know if this is already well-known:

When importing a decoder file that includes other xml files, DecoderPro will only import the main file into the user preferences and reference the includes from the installation directory. I tried it with ESU_LokSound5.xml and the "Import Decoder File" menu entry from the "File" menu in the Roster view

Trying to import the partial files fails with "does not contain a decoder definition".

So there seems to be no way to import decoder definitions with includes at the moment, and therefore also no way to test changes to included xml files without moving them to the installation directory.

Heiko

danielb987, 2023-11-15:
One thing to consider when adding this feature:

If decoder file a.xml includes file b.xml and decoder file c.xml also includes a file b.xml, and both a.xml and c.xml are imported, what to do then with b.xml?

My concern is if you import a.xml and as a result also import b.xml. And then after some time import c.xml and as a result import b.xml.

Will the latest b.xml be identical as the first b.xml? What if the new and the old b.xml are not identical? For example belonging to two different decoder manufacturers? Or that b.xml has got an upgrade that's not backwards compatible?

newHeiko, 2023-11-15:
Yes, that's another facet of the problem.

I don't know how to solve this, just that I stumbled over it :)

Heiko

bobjacobsen, 2023-11-15:
I can see two ways forward on this:

A) Update the import capability to accept a .zip file that has all the right pieces in all the right places.

B) Just remove the import capability

C) Wait to see what develops

That import capability was added a long time ago (2008!) when decoder files were simpler, new decoder definitions were being made more often, and JMRI test versions were significantly farther apart.  People would post a file to the jmriusers file section, and then others would have trouble figuring out how to download and install it.  Hence the import file and import URL capabilities.  

But things are very different now.  Test releases are routinely monthly and we have development releases: You don’t have to wait long for one with new definitions installed.  It’s great that people are working on new and updated definitions through GitHub, but I haven’t seen any user-created files floating around elsewhere for quite a while. And nobody seems to be complaining about how to do a manual download and install!

On the other hand, we certainly could create a smarter definition-import capability based on e.g. .zip files.  That’s not hard to code. Java’s .zip file support is quite reasonable.  But who would make up the .zip file packages?  Is that going to be a common use case?

So I lean toward (C): Wait and see.  If somebody arises who wants to distribute their own decoder definitions, rather than contribute them directly to JMRI, and those definitions are complex, then we could code something better.  In the meantime, we do minimal work by leaving this in place.

newHeiko, 2023-11-21:
I saw you did some work on the docs in #12633. Result looks good to me, thanks!

Heiko

closed 2023-11-21 by newHeiko