JMRI/JMRI #12651: Running Two Scripts in Startup - 2nd script appears to be not running

Full-Circle-RIJ, 2023-11-23:
Issue Type: __Bug__

In Preferences / Startup, I had two scripts setup to run. a) SetAllTurnoutsClosed  and b) SetTurnouts.
In SetTurnouts I changed two addresses, set to THROWN and removed the 3rd original line.
When the scripts run, the System Console says both scripte load without any errors, but the second Script just does not seem to actually do anything.
If i include the lines 'turnouts...' in the the 1st script, it works just fine.
Is the 2nd script running sequentially or concurrently? 

PanelPro  5.5.6+Re5414c68ff
Java  11.0.21

randall@randalljones.au

PanelPro 5.5.6+Re5414c68ff
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.21+9-LTS mixed mode
Windows 11 10.0 amd64

bobjacobsen, 2023-11-23:
It's a little complicated.  The scripts are run sequentially _but_ the `SetAllTurnoutsClosed` script kicks off a separate thread that works independently.  That thread closes a turnout every 0.1 seconds as it works all the way through.  That takes some time.  So the 2nd startup script runs while that thread is still setting turnouts closed.

The most straightforward fix might be to add a suitable delay between the two scripts in the startup preferences.

Full-Circle-RIJ, 2023-11-24:
Hi Bob

That was exactly the issue.  Thank you.

Would it be a good idea to add to the scripts that ‘work independently’ a note that any following scripts are not reliant on this script finishing?

With thanks
Randall

RANDALL JONES
(02) 9011 7001
(0409) 743 254
***@***.***
full-circle.com.au
PO Box 156
St IvesNSW2075
 The contents of this email (including any attachments) may be privileged, confidential or subject to Copyright.  Any confidentiality, privilege or Copyright is not waived or lost by reason of mistaken delivery or transmission of this email.  Any unauthorised use of the contents is expressly prohibited.  If you receive this email in error, please advise me by return email.  You must not use, distribute, copy, disclose, or take any actions, or rely on this email or its contents.
 
Viruses and Attachments:  Nothing in this message is intended to convey, represent or warrant that files attached to this email are free from computer viruses or other defects.  Any attached files are provided and may only be used on the basis that the user assumes all responsibility for virus checking any attachments.  The sender of this email does not assume any responsibility for loss, damage or consequence resulting directly or indirectly from use of the attached files.  Liability is limited as to the cost of resupply of the attached files. Thank You.

bobjacobsen, 2023-11-24:
> Would it be a good idea to add to the scripts that ‘work independently’ a note that any following scripts are not reliant on this script finishing?

Thanks for the suggestion.  I’ve done that in PR #12653

KenC57, 2023-11-24:
For scripts that run once, I've found calling them from a small script
instead of each being called in the startup seems to give the sequence
needed. This main script is called from startup but it calls one script then
the next and the next. If a script isn't spawning a process, it seems to
work.

Might it be worthwhile to put up a code example of how to sequence scripts
via internal sensors? I'm thinking a code snip you would add to the start of
a script that waits for a sensor before running. Another code snip that is
run at the end of a script to set a sensor. If you needed things to be
sequential this would be a fix so one waits for the next.

Or is there some other way that we could give users who need this sort of
thing?

bobjacobsen, 2023-11-24:
> For scripts that run once, I've found calling them from a small script
> instead of each being called in the startup seems to give the sequence
> needed. This main script is called from startup but it [...]

Not sure what the utility of that would be.

Most people want JMRI to move on while the script is e.g. setting turnouts, so the scripts are coded to do that.  I don’t think it would be a net improvement to force JMRI to wait while that’s going on.

If we do want a script to stay active until it’s complete, i.e. not fork something off, it should just be written as inline code instead of using a separate thread.

closed 2023-12-05 by bobjacobsen