JMRI/JMRI #12654: Support for TAMS LD-G-43

poldy79, 2023-11-25:
i am super new to digital model railway, managed to setup a dcc-ex station and to convert a loco to LD-W-42 and LD-G-43 and controlling it via JMRI. Cool projects. 

It seems that LD-G-43 is not yet supported by JMRI. Perhaps someone has already done some work for this. If not I will try to get into this. Has anyone else a loco with an LD-G-43?

Some links: [Overview](https://tams-online.de/epages/642f1858-c39b-4b7d-af86-f6a1feaca0e4.sf/de_DE/?ObjectPath=/Shops/642f1858-c39b-4b7d-af86-f6a1feaca0e4/Products/41-0443x) and [Manual](https://tams-online.de/WebRoot/Store11/Shops/642f1858-c39b-4b7d-af86-f6a1feaca0e4/MediaGallery/15_Download/Anleitungen/41/LD-G-43_Manual_EN.pdf)
(the manual is also available in other languages)

bobjacobsen, 2023-12-04:
@poldy79 Sorry for the delay, I was asking around.  Nobody seems to have started on this.  It would be great if you wanted to take it on.  I can help a bit.  Is the LD-G-43 similar to any of the other TAMS decoders?

closed 2024-03-18 by github-actions[bot]

PMLehmann, 2025-03-23:
Hi, same problem, LD-G-43 does not seem to be supported. Any plans on this? What are the alternatives? Can I use LD-G-42?

poldy79, 2025-03-23:
Had a short talk to the TAMS developer back in that time. He was not into DCC-EX and had also no plans. So no direct support from that side. I had yet no motivation…

About 42-Model… I think there is a list of supported controllers in the software. I do not own a LD-G-42.