JMRI/JMRI #12658: Support for PIKO SmartControl WLAN

Mafreke, 2023-11-26:
I don't know if there is support for the new PIKO SmartControl WLAN. 
if so, which system do I have to select and how do I connect it to JMRI?

if not, can someone please make it compatible? I don't have enough coding experience to make it work.

hoping for an answer.

With kind regards

bobjacobsen, 2023-11-26:
Is there any technical information about the PIKO SmartControl WLAN anywhere?  Specifically information on how (or even whether) a computer can be hooked to it?  I didn't find much with a Google search.

Mafreke, 2023-11-28:
I don't think so, there is an usb c port and if I connect my pc to it, then it shows a serial port.
I opened it up and i saw there is a esp32-Wroom inside. there's also a loconet connector but my guess is it's only for booster functionality (which is not supported right now I think) and is not a full fledged loconet port.

bobjacobsen, 2023-12-02:
I've been trying to get some more information on this.  We need to know what commands to send over the serial port, what replies to expect, etc.  Apparently that information isn't available yet, but is expected once the units are in wider distribution. Let's leave this item open in hopes that more information shows up somewhere.

Nobi322, 2024-01-22:
I'm still interested in this integration. Maybe we can sniff traffic when control unit works in main WiFi network?

bobjacobsen, 2024-01-22:
I'll leave this open.  There's not much we can do without at least a little bit of documentation.  These protocols are typically complex enough that it's hard to figure them out without that.

But we can try. If you'd like to gather the data, I can take a look at it.  Start with simple things, like reading CV1, CV2 and CV3.

closed 2024-04-22 by github-actions[bot]