JMRI/JMRI #12661: NCE Clock monitor doesn't report errors

KenC57, 2023-11-26:
Pre 5.7.1 doesn't detect out of range values. Will be part of 5.7.1 (PH5 support) soon. But it still needs a reporting method. I put in logging of error messages when something is out of range and rejected. I'll add a popup later. This is my reminder to do that. I've got code for other things, so please, nobody try fixing this right now.

KenC57, 2023-12-10:
The code preventing using out of range values is part of pull #12704 but adding a pop up for the user to know, instead of relying on a console message will come a bit later in a different pull.

KenC57, 2023-12-21:
Submitted pull #12747 that covers this issue.

bobjacobsen, 2023-12-24:
@KenC57 Is this ready to close?

closed 2023-12-24 by KenC57