JMRI/JMRI #12681: Decoder list in Decoder Pro

Firstone18, 2023-12-01:
I have recently fitted Gaugemaster decoders but these do not show up in the decoder list. I can see from the info included with the decoder they seem to be Digitrax decoders; also the CV8 values is 129 which is Digitrax.
Will Gaugemaster be added to the list sometime?
All work OK when selecting a Digitrax DZ series decoder so i can still continue using this method.
Thanks.

Firstone18, 2023-12-01:
> @firestone18 Could you read the value from CV7 and post it?  Thanks.

The value was 0 not sure if it helps?

bobjacobsen, 2023-12-02:
> The value was 0 not sure if it helps?

The combination of CV7=0 and the Digitrax code in CV129 is somewhat unfortunate.  That makes it hard to separate Gaugemaster from Digitrax decoders.  Which DZ series decoder is the closest match?  What is the exact model of your decoder?  I could (perhaps) add it as a Gaugemaster alias under the Digitrax family.

closed 2024-02-26 by github-actions[bot]