JMRI/JMRI #12692: WIndows Are Created Off Screen on R-Pi

crosland, 2023-12-06:
5.5.7 on R-Pi Bullseye

I have seen this before, with other connections/window types, but assumed I had accidentally moved a window.

It happens with a fresh LCC profile.

DP3 opens at the top left of the screen.
The LCC Network Tree opens slightly down and to the right.
The first module Configure window opens slightly further down and to the right.
The second configure module window initially opens slightly further down and to the right of the first, whilst the CDI is loaded.
When the final display is constructed, the window moves up and the title bar is obscured by the desktop Task Bar and it is not possible to access the controls to close or move the window. See attached screebshot.
A third configure window is displayed over the top of the second, with no offset.

Only the DP3 and Network Tree window positions are saved in the user-interface.xml, so it is not possible to move the Configure windows for subsequent sessions.

[image]

On R-Pi I think allowance needs to be made for the task bar which is at the top of the screen (the DP3 window does this by accident or design). It may be sufficient if second and subsequent Configure windows were offset down and right from the first.

crosland, 2023-12-07:
> From "Bob Jacobsen" ***@***.***>
> To "JMRI/JMRI" ***@***.***>
> Cc "Andrew Crosland" ***@***.***>; "Author"
> Date 06/12/2023 14:09:46
> Subject Re: [JMRI/JMRI] WIndows Are Created Off Screen on R-Pi (Issue
> #12692)

Here thay are,

Andrew

os.name: Linux
os.arch: arm
os.version: 6.1.21-v8+
Screen size h:1080, w:1920 Inset t:36, b:0; l:0, r:0
Screen resolution: 96
Environment max bounds: 
java.awt.Rectangle[x=0,y=36,width=1920,height=1044]
Device: :0.0 bounds = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
Device: :0.0 insets = java.awt.Insets[top=36,left=0,bottom=0,right=0]
JmriInsets t:36, b:0; l:0, r:0

bobjacobsen, 2023-12-07:
@crosland Thank you!

I see the problem, but I'm not sure how to fix it. There are a couple possibilities.

What's happening is that the window is initially being positioned down and to the left of the prior window OK. Then, when the data comes in, the window wants to be large enough to fit it.  So the code is expanding it to that.  But then `JmriJFrame` discovers that it doesn't fix the window any more.  So, trying to be helpful, it moves it to (what it thinks is) the top of the window and then resizes it to fit.

The problem is that what it thinks is the top of the window is too high.

A) One way to fix it is to _not_ allow it to move the window up from its prior location: Just make it as big as possible given that starting point.  This will make the windows a bit smaller, depending on how many have moved down and to the left.  But you can always resize it.

B Another approach, which is harder, is to limit the "move up" to take into account of the menu offset, much like is done with the initial starting position.  The problem is that the code that's doing the moving is acting later, when the offsets are no longer (easily) available.  

Do you have a preference?  Or perhaps another approach to how the Really Big window should be handled?

dsand47, 2023-12-07:
Bob,

Not a direct solution, but I put the Pi taskbar on the bottom of the screen.  This is set using "Desktop Preferences".

Dave Sand

crosland, 2023-12-07:
Your description didn't quite ring true so I looked at this again.

The configure windows, in their final state, are drawn with all of the tabs collapsed and the vertical height of the final state of the window I am having an issue with would easily fit the screen. Even then, It is more blank grey area, than useful info, and could be made even smaller. I have included a full height screen show to show this.

This time I noticed that a larger window is displayed very briefly. I think the window size, and where to place it, is being determined from an intermediate, larger, version of the window. It is then re-drawn in a smaller form.

This leads to a further question: Why do we (or is it just me?) get the large blank grey section between the module specific segments and the Sensor/Turnout creation? My CDI is based on the Arduino single thread example. Perhaps this should have it's own issue.

[image]

crosland, 2023-12-07:
PS I will be away for the weekend after Friday lunchtime (UK time).

bobjacobsen, 2023-12-11:
I’m not sure that it will entirely fix this, but PR 12709 (https://github.com/JMRI/JMRI/pull/12709) should make some progress on this.  It fixes how JMRI is avoiding overlaps when windows _without_ stored positions are opened.  It should be in a development release by tomorrow evening.

In the meantime, could you try deleting your user-interface.xml file to see if that problematic window is being positioned there due to some preference setting?

Thanks and sorry for the trouble.

Bob
—
Bob Jacobsen
***@***.***

crosland, 2023-12-11:
Unfortunately, it doesn't fix the issue.

I also spotted something that I didn't spot in my original description, The first module configure window was only displayed "correctly" as it is a small window when fully populated. If I open one of the other modules first, then the display is too far up the screen.

With the new files from the PR the small window is now drawn with a left and upward shift, compared to the original test, so it completely obscures the network tree (it used to be right and down from the network tree top left corner). The larger module configurations still go off the top of the screen behind the task bar. If I open a further large configuration it shifts to the left, not the right, and is slightly taller. In the screenshot you can just about make out the DP3 window. I've given the first config focus, so it is on the top, the second is the one behind and to the left.

I hope this makes sense.

[image]

dsand47, 2023-12-11:
Maybe the Raspberry Pi window manager/Java is not reporting the "available" window space correctly.  

I always have the task bar at the bottom.

Dave Sand

Screenshot.2023-12-11.170638.png (view on web) <https://github.com/JMRI/JMRI/assets/13852807/3da821f3-8b3b-43b0-88e9-c6efbfe531eb>

crosland, 2023-12-12:
No need to apologise :)

Here is the log. I opened the small config that definitely fits the screen, followed by the larger one, that also fits but is badly positioned.

settings.dir assumed to be C:\Users\andre/JMRI
ant -f C:\\JMRI_PR12709 decoderpro-remote
decoderpro-remote:
settings.dir assumed to be C:\Users\andre/JMRI
init:
copyfiles:
Copying 7 files to C:\JMRI_PR12709\target\classes
jjtree:
javacc:
update-template-code:
Copying 1 file to C:\JMRI_PR12709\target\classes
compile-generated-source:
compile:
Compiling 9 source files to C:\JMRI_PR12709\target\classes
debug:
jar:
Building jar: C:\JMRI_PR12709\jmri.jar
run with passwd
Connecting to sprog-pi4:22
done.
Connecting to sprog-pi4:22
cmd : export DISPLAY=:0; cd '/home/pi/JMRI'; /home/pi/JMRI/DecoderPro 
11:03:01,055 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
11:03:01,160 apps.util.Log4JUtil                   INFO  - This log is stored in file: /home/pi/.jmri/log/session.log [main]
11:03:01,161 apps.util.Log4JUtil                   INFO  - This log is appended to file: /home/pi/.jmri/log/messages.log [main]
11:03:01,455 apps.AppsBase                         INFO  - DecoderPro version 5.5.8plus+andrew+20231212T1101Z+Raa75902 starts under Java 11.0.18 on Linux arm v6.1.21-v8+ at Tue Dec 12 11:03:01 GMT 2023 [main]
11:03:15,209 jmri.profile.ProfileManagerDialog     INFO  - Automatically starting with profile DP3_PILCC.3f543032 after timeout. [AWT-EventQueue-0]
11:03:15,262 apps.gui3.Apps3                       INFO  - Starting with profile DP3_PILCC.3f543032 [main]
11:03:15,612 jmri.util.node.NodeIdentity           INFO  - Using 206e6304-2880-4acc-ad3d-ed3477716d2c as the JMRI storage identity for profile id 3f543032 [AWT-EventQueue-0]
11:03:15,814 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "CAN" [main]
11:03:16,297 ers.gridconnect.GcSerialDriverAdapter INFO  - ttyAMA0 port opened at 460800 baud, sees  DTR: true RTS: true DSR: false CTS: true  CD: false [main]
11:03:17,009 ri.swing.JmriJTablePersistenceManager INFO  - Table preferences not found.
This is expected on the first time the "DP3 PI-LCC" profile is used on this computer. [main]
11:03:18,220 jmri.jmrit.roster.Roster              INFO  - Reading roster file with rootFromName(/home/pi/.jmri/DP3_PILCC.jmri/roster.xml) [main]
11:03:18,384 jmri.util.FileUtilSupport             INFO  - File path program: is /home/pi/JMRI/ [main]
11:03:18,385 jmri.util.FileUtilSupport             INFO  - File path preference: is /home/pi/.jmri/DP3_PILCC.jmri/ [main]
11:03:18,386 jmri.util.FileUtilSupport             INFO  - File path profile: is /home/pi/.jmri/DP3_PILCC.jmri/ [main]
11:03:18,387 jmri.util.FileUtilSupport             INFO  - File path settings: is /home/pi/.jmri/ [main]
11:03:18,387 jmri.util.FileUtilSupport             INFO  - File path home: is /home/pi/ [main]
11:03:18,388 jmri.util.FileUtilSupport             INFO  - File path scripts: is /home/pi/JMRI/jython/ [main]
11:03:19,707 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen of "DecoderPro" starts with maximum size java.awt.Dimension[width=1920,height=1044] [main]
11:03:19,710 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with preferred height 600 width 1024 [main]
11:03:19,711 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with location 0,36 [main]
11:03:19,712 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with insets 5,25 [main]
11:03:19,714 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen normalises origin to 0, 36 [main]
11:03:19,715 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen sets height 600 width 1024 position 0,36 [main]
11:03:19,859 mri.jmrit.throttle.PowerManagerButton INFO  - No power manager instance found, panel not active [main]
11:22:44,352 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen of "LCC Network Tree" starts with maximum size java.awt.Dimension[width=1920,height=1044] [AWT-EventQueue-0]
11:22:44,353 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with preferred height 751 width 502 [AWT-EventQueue-0]
11:22:44,354 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with location 30,66 [AWT-EventQueue-0]
11:22:44,355 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with insets 5,25 [AWT-EventQueue-0]
11:22:44,355 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen normalises origin to 30, 66 [AWT-EventQueue-0]
11:22:44,356 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen sets height 751 width 502 position 30,66 [AWT-EventQueue-0]
11:22:49,399 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen of "Configure SPROG DCC Ltd - PI-LCC (02.01.2C.04.00.00)" starts with maximum size java.awt.Dimension[width=1920,height=1044] [AWT-EventQueue-0]
11:22:49,401 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with preferred height 210 width 574 [AWT-EventQueue-0]
11:22:49,402 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with location 25,66 [AWT-EventQueue-0]
11:22:49,403 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen starts with insets 5,25 [AWT-EventQueue-0]
11:22:49,404 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen normalises origin to 25, 66 [AWT-EventQueue-0]
[... cut ...]
11:23:12,235 jmri.util.JmriJFrame                  INFO  - reSizeToFitScreen sets "Configure SPROG DCC Ltd - SERVOIO-LCC (02.01.2C.01.05.00)" Y to minimum 30 [AWT-EventQueue-0]
11:23:12,236 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen during Y calculation sets location 50, 30 [AWT-EventQueue-0]
11:23:12,237 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen sets "Configure SPROG DCC Ltd - SERVOIO-LCC (02.01.2C.01.05.00)" height to 948 [AWT-EventQueue-0]
11:23:12,240 jmri.util.JmriJFrame                  INFO  - reSizeToFitOnScreen sets height 948 width 905 position 50,30 [AWT-EventQueue-0]

bobjacobsen, 2023-12-12:
From this, it appears that putting the windows up under the main Linux bar is either a mystery, or a problem with how Java and your window manager are interacting.

Java is supposed to position its windows relative to the _usable_ area of the screen.  0,0 is supposed to be the upper-left corner of the usable space, below any top space that's used by the OS.  

From the debug info, all those windows are being positioned at positive X and Y locations, which shouldn't put them under the Linux bar. They should even have a little space between the bar and the top of the windows.  None of them have negative Y values that would slide them up and under.  Yet they are!

(N.B. If the window positions and sizes have been manually set, e.g. are in the user-interface.xml file, that happens separately from this logging. That's why I suggested deleting the active version of that file)

I'm not sure where to go from here.  I could add an always-present offset that would push down all windows on Linux, but that would leave unusable space at the top for people who don't have the Linux bar there.

crosland, 2023-12-13:
OK, it seems putting the task bar at the bottom may be the only 
solution.

I recall reading that the R-Pi task bar is an application and they had 
to do quite a lot of work to port it to Wayland in the new Bookworm OS. 
I don't know if that is any different to other Linux distros, but the 
ones I have seen all seem to have the task bar at the bottom of the 
screen. I guess the R-Pi foundation wanted to be different.

I'll post your comments (anonymously) on the R-Pi forum and see if there 
any other suggestions.

I will also test on Bookworm and see if the behaviour is any different.

Andrew

Cc "Andrew Crosland" ***@***.***>; "Mention" 
Date 12/12/2023 20:39:06

crosland, 2024-01-29:
Closed, Workaround is to move the taskbar or move the window after it's created.

closed 2024-01-29 by crosland