JMRI/JMRI #12713: Feature Request: Documentation in CDI

crosland, 2023-12-13:
I successfully created an "empty" segment in a nodes xml, to which I can add text using the description tag, e.g.,

```
  <segment origin='130' space='253'>\
    <name>Documentation</name>\
      <description>This is documentation! \
      </description>\
  </segment>\
  <segment origin='130' space='253'>\

```

Would it be possible to embed URLs to create clickable links to the manufacturers documentation for a node?

bobjacobsen, 2023-12-13:
Could you please try wrapping the entry in an `<html>` tag?

```
      <description>This is documentation! <html><a href="somewhere">My link</a></html>
      </description>\
```

That might work because of the way Java handles text, but I'm not entirely sure that it won't get stripped somewhere along the pipeline. 

Other apps might not be able to handle it, though. So perhaps we need to discuss a standard-defined way to do this.

crosland, 2023-12-13:
I had tried various ways of embedding the URL but not using a html tag. Unfortunately, this does not work, either. I had to escape the quotes for the XC8 compiler as the xml is defined as a string in a char[].

[code]
<html><a href=\"https:\\www.sprog-dcc.co.uk\">My link</a></html> click the link!
[/code]

Nothing within the html tags is displayed.

Maybe the text field used for displaying the CDI does not support html.

bobjacobsen, 2023-12-13:
It looks like the contained `<A>` element is being stripped in the underlying OpenLCB-Java and OpenLCB-Swift libraries.  There are probably issues downstream of that too.  

From an implementation point of view, it might be better to carry this in the CDI some other way. I've tried to start a [https://groups.io/g/openlcb/message/15641](discussion on the OpenLCB list).

danielb987, 2023-12-13:
Some web forums uses `[link=http://some_url]Some text[/link]` and `[link]http://some_url[/link]` instead of using `<a>` tags. They do it for security reasons, but the same principle can be used here.

crosland, 2024-01-29:
Doesn't seem like this has much interest.

closed 2024-01-29 by crosland