JMRI/JMRI #12723: "Where Used" tool missing some uses of Reporters

jerryg2003, 2023-12-15:
I set up reporters linked to blocks, referenced in LogixNGs, and on a CPE panel.  The "where used" tool only finds them in the block reference:

From my panel file:
Initial definition:
	Line  1022:       <systemName>MRE32-RFID02/L</systemName>
Associated with a block:
	Line  1529:       <reporter systemName="MRE32-RFID02/L" useCurrent="no" />
In a LogixNG expression (under ExpressionReporter namedbean):
	Line  8535:         <name>MRE32-RFID02/L</name>
On a CPE panel (the first item listed on that panel):
	Line 17754:     <reportericon reporter="MRE32-RFID02/L" x="200" y="80" level="5" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" size="11" style="0" red="0" green="0" blue="0" hasBackground="no" justification="centre" class="jmri.jmrit.display.configurexml.ReporterIconXml" />

As found by Where Used:
`Reporter: MRE32-RFID02/L	
		Listener count: 5	

	Block references	
		Siding #1 (IB5)	
`

Is the tool missing these additional references?

dsand47, 2023-12-15:
Please upload a sample layout data file.

jerryg2003, 2023-12-15:
@dsand47 I've paired out as much as I could readily do.  Still has all the items from above.  [Added .txt to upload]

[TEST Where Used issue1.xml.txt](https://github.com/JMRI/JMRI/files/13688158/TEST.Where.Used.issue1.xml.txt)

dsand47, 2023-12-15:
Support for Reporters needed a couple of minor changes.
```

		Reporter: MRE32-RFID02/L	
		Listener count: 4	

	Block references	
		Siding #1 (IB5)	

	LogixNG references	
		OCC: Set/reset Siding #1 (IQ:AUTO:0059), IQC:AUTO:0105, expression: Reporter MRE32-RFID02/L Current Report is not null	

	Panel references	
		Test Loco Tracking Editor :: ReporterIcon :: x=200, y=80	
```

jerryg2003, 2023-12-16:
Thanks for developing a great tool for JMRI!

closed 2023-12-17 by bobjacobsen