JMRI/JMRI #12725: Websockets: no message when turnout changed

wijnanb, 2023-12-15:
I am writing a Flutter app that works with the websocket of JMRI.

Now I can connect to the websocket. I receive the hello message and can do the heartbeat.

But when I change a turnout or a light, it will not send a message over the websocket.

Do I need to subscribe to a channel or something to receive the messages? 
I cannot find much information about.
Or do I need to implement a Jython script on the JMRI server to send out the messages myself? 

```
flutter: WebSocket message: {"type":"hello","data":{"JMRI":"5.4+R9005ecb4ac","json":"5.4.0","version":"v5","heartbeat":13500,"railroad":"My Railroad","node":"jmri-909C4AB89EEA-3ed37e8c","activeProfile":"Simulation"}}
flutter: Websocket send heartbeat
flutter: WebSocket message: {"type":"pong"}
flutter: Websocket send heartbeat
flutter: WebSocket message: {"type":"pong"}
```

mstevetodd, 2023-12-16:
Hi Bert,
You need to send a websockets message from your client for each item you're interested in.
The /table JavaScript does this, so you can see the format needed.
I'd also recommend using the /json page as a testing tool.
--SteveT

mstevetodd, 2023-12-16:
Here's an example you can play with here: http://jmri.mstevetodd.com:12080/json/
Send 
`{"type":"turnout","data":{"name":"LT105"}}`
Then open another window to http://jmri.mstevetodd.com:12080/tables/turnout and toggle the state of LT105.

wijnanb, 2023-12-18:
Thanks @mstevetodd Now I managed to have realtime update across multiple devices.
I never figured that out myself from the documentation.

closed 2023-12-18 by wijnanb

mstevetodd, 2023-12-18:
> I never figured that out myself from the documentation.

Understood and agree. Don't get me started. I had additional documentation and especially numerous examples removed by another developer. He felt that my documentation referred only to the "normal" JMRI and did not account for developers overriding the normal JMRI functions. He added the Notes section here: https://www.jmri.org/JavaDoc/doc/jmri/server/json/package-summary.html as a replacement for my stuff.

danielb987, 2023-12-18:
@mstevetodd 
Do you know which PR removed your documentation? Could it be PR #6777 ?

rhwood who created that PR has left JMRI so my suggestion is that if you want, that you restore your documentation.

mstevetodd, 2023-12-18:
Yes, that definitely includes the bulk of what I recall being removed. 
I will take a look at adding some of that back. Thanks for finding the PR, that'll make it easier.