JMRI/JMRI #12729: Enginedriver cannot connect to WiThrottle

LinuxLover9, 2023-12-17:
I've recently installed Decoder Pro 5.4 on Ubuntu 20.04.5 with java 17.0.9
I can control my Digitrax Zephyr without issue.
However when I enable the WiThrottle, it will not connect with Enginedriver v2.35.173.

I see my server as discovered with it's proper IPv4 address, but a netstat in Ubuntu only shows it listening on a IPv6 address. Therefore Enginedriver times out, when trying to connect.
Even when I disable broadcasting IPv6 in the preferences, it still only listens on the IPv6 address.

Searching the issues list, this only has come up in the distant past, so it does not seem to be a current issue for most.

Any suggestions on how to fix this?

Thank you.

mstevetodd, 2023-12-17:
The broadcast is a completely different mechanism, it's an outbound communication.
Most of the time, this problem is "firewall". Disable all firewalls on your JMRI computer and try again. If that works, add an exception for the ports you're using and turn the firewall back on.

LinuxLover9, 2023-12-17:
I know about networks and it needs to listen on the proper port!
```
localhost:~$ netstat -atl | grep LIST
tcp        0      0 localhost:5550          0.0.0.0:*               LISTEN     
tcp6       0      0 [::]:12090              [::]:*                  LISTEN 
```
I miss a line with `tcp 0.0.0.0:12090 0.0.0.0:* LISTEN` to indicate it's listening on IPv4.

LinuxLover9, 2023-12-17:
One step forward.
Had to force java to use the IPv4 stack in .profile or .bashrc:
```
# make sure java uses IPv4
export _JAVA_OPTIONS="-Djava.net.preferIPv4Stack=true"
```
Now I see what I expected
```
localhost:~$ sudo netstat -atlp | grep LIST
tcp        0      0 localhost:5550          0.0.0.0:*               LISTEN      1747/patchpaneld    
tcp        0      0 0.0.0.0:12090           0.0.0.0:*               LISTEN      2194/java
```
and I don't have the fw enabled
```
localhost:~$ sudo ufw status
Status: inactive
```
But Engine driver still times out... :(
Now what?

mstevetodd, 2023-12-17:
LinuxLover9,
I ran your command on my Ubuntu machine, which definitely works with EngineDriver, and I get similar results as you: The tcp connection is NOT listed, but the tcp6 connection IS listed. (Note: I do not have preferIPv4Stack set).

Do you have something blocked in iptables?

Also, please check your JMRI System Log for errors, and paste the actual error message text you're getting in EngineDriver, using View Log.

--SteveT

closed 2024-03-11 by github-actions[bot]