JMRI/JMRI #12734: Issue with Block value not getting updated from Reporter

jerryg2003, 2023-12-18:
JMRI 5.5.7, Windows 11, Azul Java 11.0.16.1:

Scenario: Using MQTT Reporters to report RFID tag IDs.  All works as expected EXCEPT the associated Block value is not updated when same Report (RFID tag) is seen again after being cleared (but still in the Last Report field).

I was testing my RFID readers by running a loco back and forth over an RFID reader approx. very 10 seconds.  I have a Block linked to the MQTT Reporter and also a sensor for block occupancy.  [My microprocessor publishes an MQTT Sensor ACTIVE message and MQTT Reporter message with the ID Tag, and 2 seconds later publishes a MQTT Sensor INACTIVE and a Reporter message with a blank value.] 

The first time, the Current Report and Last Report and Block value are filled in correctly and the block shows occupied.  Second and subsequent times, the Current Report and Last Report are filled and the block shows occupied, but the Block value remains blank.  Two seconds later, the Reporter current value goes blank (I publish a blank message) and Last Report stays as expected and the block goes unoccupied.  [Additional weirdness: sometimes, between the 5th and 15th time I try the same report, the block value will fill in! Haven't been able to narrow it down more than that.]

If I send a different loco over the RFID reader, and then the first loco, all works as expected. So somehow the issue of the Block value not getting filled in seems to be related to the Reporter current value going back and forth between a particular value and blank (although why it sometimes work after many tries is beyond me).  If the Reporter gets a different value, all works as expected.

bobjacobsen, 2023-12-18:
From just a quick look at the code, it appears that it's assuming that when the empty report arrives, the block is empty.  Is that what you intended?

jerryg2003, 2023-12-18:
Not sure about “intentions,” but I see in my code that I publish the MQTT message setting the block occupancy sensor to INACTIVE after (immediately after - next line of code)  I send the Reporter clear.  Perhaps a timing issue in JMRI?  Let me do an experiment changing the order of the messages in my ESP32 code and see if that makes a difference.  Thanks.

Pugwash1, 2023-12-18:
@jerryg2003 
The block value is  filled in by the block tracking code, and can be updated quite a while after it goes active/inactive as its looking for an ajacent(?) (next to) block going inactive. We should put in an option to turn block tracking off.

jerryg2003, 2023-12-18:
@Pugwash1 Is that true even if I associate a Reporter explicitly with the block? I'm not counting on the block tracking to fill the value in, but rather the MQTT message published to that Reporter.

@bobjacobsen I changed the order of my MQTT messages to have the sensor INACTIVE before the reporter clear and that improved the situation but didn't totally cure it (now I'm really confused)...  Here are examples of my MQTT messages and I can't see any difference between a set where the block value is correctly populated and one where it isn't.  In this example, I had a string of five times when the value was populated (I'm only showing messages from the last one) and then the next one where it didn't (I also show these messages).  I can't see any difference in these sets of messages at all.  So it seems like the behavior must be some timing thing that it happening in JMRI itself.

One other point that may or may not be relevant:  I have user names in my IDTag table associated with each tag.  So while my MQTT message has the raw IDTag name, it is the user name that is populated in the block value (when it is) as well as in the Reporter report.
`
Example where Block value IS populated:
2023-12-18T12:15:52-0500    jtrains/jRecv/reporter/E32-RFID02/L 0 0           045b89d2226d81
2023-12-18T12:15:53-0500         jtains/jRecv/sens/E32-RFID02/L 0 0                   ACTIVE
I fill in a memory for use elsewhere...
2023-12-18T12:15:53-0500         jtrains/jRecv/mem/E32-RFID02/L 0 0           045b89d2226d81
2 seconds later...
2023-12-18T12:15:54-0500        jtrains/jRecv/sens/E32-RFID02/L 0 0                 INACTIVE
2023-12-18T12:15:54-0500    jtrains/jRecv/reporter/E32-RFID02/L 0 0

Heartbeat I send out every five periodically...
2023-12-18T12:15:56-0500       jtrains/jRecv/sens/E32-RFID02/HB 0 0                   ACTIVE

Then, the very next time the RFID tag is read, the Block value is NOT populated:
2023-12-18T12:15:57-0500    jtrains/jRecv/reporter/E32-RFID02/L 0 0           045b89d2226d81
2023-12-18T12:15:57-0500        jtrains/jRecv/sens/E32-RFID02/L 0 0                   ACTIVE
2023-12-18T12:15:57-0500         jtrains/jRecv/mem/E32-RFID02/L 0 0           045b89d2226d81
2023-12-18T12:15:59-0500        jtrains/jRecv/sens/E32-RFID02/L 0 0                 INACTIVE
2023-12-18T12:15:59-0500    jtrains/jRecv/reporter/E32-RFID02/L 0 0
2023-12-18T12:16:01-0500       jtrains/jRecv/sens/E32-RFID02/HB 0 0                 INACTIVE
2023-12-18T12:16:06-0500       jtrains/jRecv/sens/E32-RFID02/HB 0 0                   ACTIVE
`

Pugwash1, 2023-12-18:
@jerryg2003 
Yep true. Block.java
`/**
     * Handles Block sensor going ACTIVE: this block is now occupied, figure out
     * from who and copy their value.
     */
    public void goingActive() {`

Pugwash1, 2023-12-18:
@jerryg2003 
The only question is: 
1. Do we just make a public in block manager that can be set off by a script, do we allow overides by another public in the block to set off or on individually
2. Do we have to give it a gui?

If 1 I will happily do it, if 2 I am sure someone else can.

-

dsand47, 2023-12-18:
How about adding an entry to the Block Values menu?
Add a separator and "Disable block tracking".

jerryg2003, 2023-12-18:
Let me try to describe the actual situation on my layout and see how this might work.  I have a number of blocks which have occupancy sensors and a few (spaced around the layout) that don’t.  In those, I have put RFID readers.  I would like to have a train go across an RFID Reader and have its ID put in that block’s value (I assign a “logical” sensor to reader which I set ACTIVE when the reader reads a tag).  [I use MQTT and MQTT Reporters, but other types should work the same way.]

As that train now traverses other blocks, I would like the value to follow it (i.e. block tracking).  I know this may or may not work perfectly in the real world (sensor failures, etc.).  So I have other RFID readers around the layout and when the train crosses one, that block gets its value.  The RFID Reader value would override any block tracking that comes from the adjacent prior block.  [Wouldn’t transponding also work the same way with LocoNet Reporters?]. 

Other scenarios for using RFID might not require block tracking (such as identifying freight cars in a hidden yard) but mine does.  I just need JMRI to reliably put the report into the block value. Right now, if I only have one train going around, it wouldn’t do that because sometimes it would get the report, show the reader’s block as occupied but then not put the report value into the block value.

jerryg2003, 2023-12-18:
I also tried publishing the sensor ACTIVE before publishing the Report with the TagID to see what that would do.  The attached video shows that it still only sometimes populates the Block Value (for Siding #1, highlighted in the block table) although the block shows Occupied when it should.  The small CPE panel is displaying the current Report field for Reader L (first on the Reporter Table window).  The partial Layout panel shows the block value between the red and blue dots.  Siding #1 is the center of the three vertical tracks and you can see it turn red.  Also shown is the ID Tag Table (first entry).  All these update as expected even though the block value does not always populate.  The right side of the screen is my MQTT message monitor.

https://github.com/JMRI/JMRI/assets/42826502/95ef40ff-f47e-41d9-acd0-bfdc4f75d6f4

jerryg2003, 2023-12-18:
To summarize, and perhaps naively based on the Block Table user documentation (“Reporter: The Reporter assigned to this Block. Puts the Reporter Value into the Block value.”), my expectation as a user is that if a reporter is assigned to a block, its value will be placed in the block value.  This happens now when a new Current Report (blank or not) not equal to the Last Report appears, and sometimes when new Current Report is the same as the Last Report.  My mental model as a user is that “sometimes” should not be part of the behavior.  Thanks.

jerryg2003, 2023-12-23:
FYI:  I found a work around: NOT specify the Reporter in the Block entry.  Instead, create a LogixNG to populate the Block Value when the Reporter changes.  Works find for my purposes, but still leaves the mystery of why specifying the Reporter in the Block Table entry works sometimes but not all the time.

jerryg2003, 2023-12-27:
Further analysis shows that block value is populated (and cleared) as expected when the "Current Report?" option is selected for the Block Reporter.  However, it does NOT appear to work correctly (except sporadically, which is most confusing) when "Current Report?" is unselected, based on the documentation on the web site:

[https://www.jmri.org/help/en/package/jmri/jmrit/beantable/BlockTable.shtml](url)
`A Reporter, however, can tell us one of the following:

'Hey, I'm seeing something right now - this is it'
'Hmm, I'm not seeing anything right now'
'Hey, I saw something a while back - this was it'
'Hmm, I've not seen anything yet'
The first two are from the CurrentReport; the last two from LastReport. At start-up (assuming there is nothing within the range of the reader), both 2 and 4 are the initial state. Once something is seen, both 1 and 3 return the same result. When the something is 'out-of-range' 2 and 3 are then active. With 'Use Current?' selected, the Block value will switch between 1 and 2, whereas with 'Use Current?' unselected the Block value will switch between 3 and 4.`

jerryg2003, 2024-01-08:
@Pugwash1 Any chance we could actually talk about this or communicate in a more interactive way than this?  I'm now trying to update the JMRI web site / documentation for blocks [see my previous post in this Issue] and unable to reconcile what I see happening with what the current documentation says.  As more people start to use RFID and other Reporters, especially with microcontrollers where they are generating their own reports, we should be able to clearly explain what happens under what circumstances.  With your being able to understand the block.java code and me testing it, I think we can resolve this fairly quickly.  Thanks.

jerryg2003, 2024-01-11:
[TLDR: There is a problem with interaction between Reporters and block tracking.  @Pugwash1 is working on coming up with a solution.]

After an extensive discussion and demonstration with @Pugwash1  (thank you!) in which he also noted there were related issues using Digitrax transponding and RailCom, we seem to have hit on the underlying issue, namely interaction between Reporters and block tracking.  While we don’t have a solution yet, we can better define the various scenarios.  [In the text below, the terminology “train moves to” means the first block goes UNOCCPUIED and the second goes OCCUPIED (i.e. the associated sensor goes from ACTIVE to INACTIVE and vice versa.]

For context,  Reports are only sent to JMRI when the train and the block are both equipped with appropriate hardware, either RFID tag and reader, Digitrax transponder and decoder, RailCom transponder and decoder, etc.).  We have to account for cases where either a train or a block (or both) are not so equipped but others are.
 
(1)	Base case: Train moving from block A to B to C (assumes block occupancy determined from sensor) but train does NOT trigger a report (e.g. either the train does not have a tag or transponder, or the blocks  do not have hardware capable of generating a report).  Normal “block tracking” should work: any value in A (perhaps entered manually) should be moved to B and then to C.
(2)	Train and block B have ability to trigger a report (has RFID tag/reader or transponder/reader, etc.):  As train moves from A to B, the block value in A should be cleared but not moved to B (i.e. discarded).  The Report from B should be put in the block value for B.  As the train moves to C, B’s block value (containing the report) should be moved from B to C.
(3)	Train and ALL blocks have ability to trigger a report (has RFID tag/reader or transponder/reader, etc.):  each block independently applies the rule in (2).  The effect is that the report gets put into block value for each block.  There is no “block tracking” as such.
(4)	Some block have the ability to trigger a report but the train does not (no tag or transponder): same as case (1).  The block must have some means to set a sensor to ACTIVE so that block tracking can continue “across” this block even if the train does not trigger a report.
(5)  Some blocks and some trains have appropriate hardware to trigger a report.  Combination of block tracking and Report copying.

@Pugwash1  is going to think about what might be done easily to make all the cases work.

Thanks again for all who are participating in this discussion.

bobjacobsen, 2024-01-11:
> (2) Train and block B have ability to trigger a report (has RFID tag/reader or transponder/reader, etc.): As train moves from A to B, the block value in A should be cleared but not moved to B (i.e. [...]

I think treating this as a special case is a mistake.  JMRI doesn't know that "Train .. (has) ability to trigger a report" vs not being able to.  Instead of "not moved to B (i.e. discarded).", this case should be similar to (1), where the block tracking will do the move based on Sensor readings, _and_ the Reporter can also enter the value, which-ever comes first.

bobjacobsen, 2024-02-26:
There has recently been some discussion of this on JMRI users: https://groups.io/g/jmriusers/message/227323

jerryg2003, 2024-05-11:
Can we keep this issue open until Stephen Johnson or I have time to do some more experiments?  As far as I know, the issue still exists.