JMRI/JMRI #12745: Panel changes can create LogixNG errors.

dsand47, 2023-12-21:
The **Icon/Label** Display actions are based on **IDs** or **Classes** added to positionable objects on panels.

If the id or class name is changed, the actions fail with an error message in the system console.  The error also occurs if the text is deleted or the positionable itself has been deleted.

The current message provides no details about the missing content.

See the discussion at user group thread https://groups.io/g/jmriusers/message/224269.

danielb987, 2023-12-23:
I'm thinking of this, but I'm entirely sure on how to solve it. Something like a "canDelete" is needed.

An issue is with classes. If the user wants to rename the class of several positionables, should a warning message be shown every time? For example if he has ten positionables with that class, but only wants to change five of them.

It would be good if the user had the option to rename a class so the change was done to all the positionables using that class, as well as all the LogixNGs using that class.

dsand47, 2023-12-23:
I think there are three possible approaches.

The easiest is to improve the error message to include necessary details so that the user can fix the action.  The details need to include the parent ConditionalNG and LogixNG/Module/Clipboard.

The second approach is to search the actions when an id or class changes and present a list of affected actions.

The third approach to implement a veto process.

danielb987, 2024-01-21:
> The easiest is to improve the error message to include necessary details so that the user can fix the action. The details need to include the parent ConditionalNG and LogixNG/Module/Clipboard.

This is now implemented by #12847. It would be good if you could check the PR and see if the error messages works for you.

I haven't added an error message for the ActionLayoutTurnout. If the turnout doesn't exist when the panel is loaded, ActionLayoutTurnout will not have a turnout to act on.

dsand47, 2024-01-21:
The generated message looks good.  It provides the necessary details.

closed 2024-04-15 by github-actions[bot]