JMRI/JMRI #12754: Change in the way roster images are linked from roster entries??

philguk, 2023-12-23:
I've noticed a change in the way JMRI links roster images to their respective roster entries

In the past, individual roster entries have included something like the following

imageFilePath="preference:roster/<image filename>.JPG" iconFilePath="preference:roster/<image filename>.JPG"

This format is nicely platform agnostic

Now they look like this for Windows
imageFilePath="//<server name>/<full path>/roster/RhB 62.jpg" iconFilePath="//<server name>/<full path>/roster/RhB 62.jpg"

and this for Linux
imageFilePath="home:<mount point>/<full path>/roster/RhB 62.jpg" iconFilePath="home:<mount point>/<full path>/roster/RhB 62.jpg"

This format is definitely NOT agnostic!

I've been trying to find out when this change happened, but haven't been able to narrow it down yet, but it seems to have been like this since at least 5.3.3

If there is a change I might have made that could cause this behaviour change, PLEASE tell me!!

Thanks

bobjacobsen, 2023-12-23:
I don' think there has been change to how JMRI works in this area. At least, I don't see one in the code.

JMRI starts with a full file pathname.  It then tries to match the first part of that to the directories it knows about.  If the start of the path matches _exactly_ with the `home` path, then it will be replaced by `home:`.  Same for the `preferences` path, etc.

What is the full pathname to the icon file?

When you go to Help -> File Locations, what does it show?

philguk, 2023-12-23:
Aha!

What is the “preferences” path? Maybe I’ve set that wrong since we moved house?

I don’t see “preferences” in the list of file locations

Fingers crossed

Phil G

dsand47, 2023-12-23:
Phil,

"preference:" is the internal keyword for the user files location.  

The full details are at https://www.jmri.org/help/en/html/doc/Technical/ProfileFileStructure.shtml

Dave Sand

philguk, 2023-12-23:
Thanks Dave

This sounds really promising!

I’ll take a look later, but pretty sure that’s NOT set to where I need it to be

I guess the original assumption was that the roster would always be here. Now that it can be elsewhere, things can break if user files are not kept with the roster

Hopefully report back later…… 🤞🤞

Phil G

KenC57, 2023-12-23:
Which order does it do the matching? If it starts with the deepest path and
then works for shorter paths, it should be doing ok. But if home: was the
first check, that would match just about everything for a normal windows
user. On other systems, I'm not so sure.

dsand47, 2023-12-23:
Phil,

I did a roster test.  Since I don't have a NAS, I used my Mac as the NAS with Windows and Ubuntu connecting to the Mac using Parallels.

Here are the user files location and roster location for each OS, including the Mac.
```
User Files Location: Z:\JMRI\_Configs\RemoteFiles\
Roster Location: Z:\JMRI\_Configs\RemoteFiles\

User Files Location: /media/psf/JMRI/_Configs/RemoteFiles/
Roster Location: /media/psf/JMRI/_Configs/RemoteFiles/

User Files Location: /Users/das/JMRI/_Configs/RemoteFiles/
Roster Location: /Users/das/JMRI/_Configs/RemoteFiles/
```

Here is the image path in the roster file:
imageFilePath="preference:roster/TinyClock.png"

(I don't have any loco images)

In this test, the roster and user files location are the same, so the preference: keyword is used.

philguk, 2023-12-23:
Thanks Dave

Just did a similar test, and all is working as I thought it should!

Apart from the fact that the first time I opened the Roster ALL of my 
images went AWOL. But adding them back, whilst a bit f a pain, is not 
the end of the world

Thanks also to Bob J for getting me looking in the right place

Phil G

PS I'll close the issue I opened

--------------MXbbEgyv0Fp0mvxOIlXAw1Q7
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>Thanks Dave</p>
    <p>Just did a similar test, and all is working as I thought it
      should!</p>
    <p>Apart from the fact that the first time I opened the Roster ALL
      of my images went AWOL. But adding them back, whilst a bit f a
      pain, is not the end of the world</p>
    <p>Thanks also to Bob J for getting me looking in the right place</p>
    <p>Phil G</p>
    <p>PS I'll close the issue I opened<br>
    </p>
    <div class="moz-cite-prefix">On 23/12/2023 16:32, Dave Sand wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">Phil,</p>
      <p dir="auto">I did a roster test. Since I don't have a NAS, I
        used my Mac as the NAS with Windows and Ubuntu connecting to the
        Mac using Parallels.</p>
      <p dir="auto">Here are the user files location and roster location
        for each OS, including the Mac.</p>
      <pre class="notranslate"><code class="notranslate">User Files Location: Z:\JMRI\_Configs\RemoteFiles\
Roster Location: Z:\JMRI\_Configs\RemoteFiles\

User Files Location: /media/psf/JMRI/_Configs/RemoteFiles/
Roster Location: /media/psf/JMRI/_Configs/RemoteFiles/

User Files Location: /Users/das/JMRI/_Configs/RemoteFiles/
Roster Location: /Users/das/JMRI/_Configs/RemoteFiles/
</code></pre>
      <p dir="auto">Here is the image path in the roster file:<br>
        imageFilePath=&quot;preference:roster/TinyClock.png&quot;</p>
      <p dir="auto">(I don't have any loco images)</p>
      <p dir="auto">In this test, the roster and user files location are
        the same, so the preference: keyword is used.</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          ID: <span>&lt;JMRI/JMRI/issues/12754/1868326673</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/JMRI/JMRI/issues/12754#issuecomment-1868326673",
"url": "https://github.com/JMRI/JMRI/issues/12754#issuecomment-1868326673",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------MXbbEgyv0Fp0mvxOIlXAw1Q7--

philguk, 2023-12-23:
Issue now closed - "user error" :)

closed 2023-12-23 by philguk