JMRI/JMRI #12763: PanelPro stops unexpectedly on opening Tables

voroskoi, 2023-12-25:
Hi,

First of all: Thank You very much for working on JMRI, I have just installed it for my EX-CommandCenter and it works fine with WiThrottle server.

However I would like to follow my layout with PanelPro. I can draw the layout, but when I try to add sensors or block in `Tools->Tables` it just exits immediately.
There is not much of an error message:

```sh
jmri  master is 📦 v5.7.1-SNAPSHOT via ☕ v21.0.1 ❯ ant panelpro
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=on
Buildfile: /home/voroskoi/code/jmri/build.xml
     [echo] settings.dir assumed to be /home/voroskoi/.jmri

init:

copyfiles:

jjtree:

javacc:

update-template-code:
     [copy] Copying 1 file to /home/voroskoi/code/jmri/target/classes

compile-generated-source:

compile:

debug:

panelpro:
     [echo] settings.dir assumed to be /home/voroskoi/.jmri

runtime-library-selection:
     [echo] arch.lib.path /home/voroskoi/code/jmri/lib/linux/x86_64:/home/voroskoi/code/jmri/lib/linux

-normal-jmri-application:
     [echo] Launch normally (no debugger support)
     [java] Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=on
     [java] 18:42:16,234 apps.util.Log4JUtil                   INFO  - ****** JMRI log ******* [main]
     [java] 18:42:16,246 apps.util.Log4JUtil                   INFO  - This log is stored in file: /home/voroskoi/.jmri/log/session.log [main]
     [java] 18:42:16,246 apps.util.Log4JUtil                   INFO  - This log is appended to file: /home/voroskoi/.jmri/log/messages.log [main]
     [java] 18:42:16,257 apps.Apps                             INFO  - PanelPro version 5.7.1plus+voroskoi+20231225T1742Z+R1c2592fe06 starts under Java 21.0.1 on Linux amd64 v6.6.8-0-lts at Mon Dec 25 18:42:16 CET 2023 [main]
     [java] 18:42:16,320 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3ee154a2 [main]
     [java] 18:42:16,388 jmri.util.node.NodeIdentity           INFO  - Using 25d86af0-76c6-49ef-b51f-2beaee87c544 as the JMRI storage identity for profile id 3ee154a2 [AWT-EventQueue-0]
     [java] 18:42:16,452 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "DCC++" [main]
     [java] 18:42:16,504 jmri.jmrix.dccpp.serial.DCCppAdapter  ERROR - Serial port ttyACM0 not found [main]
     [java] 18:42:16,506 jmri.jmrix.dccpp.serial.DCCppAdapter  ERROR - failed to connect DCC++ to ttyACM0 [main]
     [java] 18:42:16,507 jmri.jmrix.ConnectionConfigManager    ERROR - Unable to create jmri.jmrix.dccpp.serial.configurexml.ConnectionConfigXml for [Element: <connection/>], load returned false [main]
     [java] 18:42:16,508 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.jmrix.ConnectionConfigManager: Unable to create several of your connections. [main]
     [java] 18:42:16,818 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.util.startup.StartupActionsManager: jmri.util.prefs.InitializationException: Unable to run startup actions due to earlier failures. [main]
     [java] 18:42:18,761 jmri.util.FileUtilSupport             INFO  - File path program: is /home/voroskoi/code/jmri/ [main]
     [java] 18:42:18,762 jmri.util.FileUtilSupport             INFO  - File path preference: is /home/voroskoi/.jmri/My_JMRI_Railroad.jmri/ [main]
     [java] 18:42:18,763 jmri.util.FileUtilSupport             INFO  - File path profile: is /home/voroskoi/.jmri/My_JMRI_Railroad.jmri/ [main]
     [java] 18:42:18,764 jmri.util.FileUtilSupport             INFO  - File path settings: is /home/voroskoi/.jmri/ [main]
     [java] 18:42:18,765 jmri.util.FileUtilSupport             INFO  - File path home: is /home/voroskoi/ [main]
     [java] 18:42:18,765 jmri.util.FileUtilSupport             INFO  - File path scripts: is /home/voroskoi/code/jmri/jython/ [main]
     [java] 18:42:19,156 jmri.util.HelpUtil                    WARN  - unable to show help page package.apps.AppConfigPanelErrorPage due to: [main]
     [java] jmri.JmriException: Failed to connect to web page. java.awt.Desktop doesn't suppport Action.BROWSE
     [java] 	at jmri.util.HelpUtil.openWebPage(HelpUtil.java:250) ~[classes/:?]
     [java] 	at jmri.util.HelpUtil.showWebPage(HelpUtil.java:226) [classes/:?]
     [java] 	at jmri.util.HelpUtil.displayHelpRef(HelpUtil.java:181) [classes/:?]
     [java] 	at apps.Apps.<init>(Apps.java:317) [classes/:?]
     [java] 	at apps.PanelPro.PanelPro.<init>(PanelPro.java:40) [classes/:?]
     [java] 	at apps.PanelPro.PanelPro.main(PanelPro.java:123) [classes/:?]
     [java] 18:42:23,793 apps.PanelPro.PanelPro                INFO  - Main initialization done [main]
     [java] terminate called after throwing an instance of 'std::system_error'
     [java]   what():  No error information
     [java] Java Result: 134

-debug-jmri-application:

-run-jmri-application:
     [echo] apps.PanelPro.PanelPro finished with return code 134
     [echo] settings.dir assumed to be /home/voroskoi/.jmri

-relaunch-jmri-application:

BUILD SUCCESSFUL
Total time: 1 minute 28 seconds
```

This is with latest master on Alpine Linux edge built with `ant debug`, but I have tried the latest prebuilt version (5.6) too, which gives the same results.

Java version info:
```
jmri  master is 📦 v5.7.1-SNAPSHOT via ☕ v21.0.1 ❯ java --version
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=on
openjdk 21.0.1 2023-10-17
OpenJDK Runtime Environment (build 21.0.1+12-alpine-r2)
OpenJDK 64-Bit Server VM (build 21.0.1+12-alpine-r2, mixed mode, sharing)
```

Sadly I am not familiar with java, so I have no idea how to get some tracing information on this. If there is something I could try, please let me know, I will do my best to figure this out. I also happy to provide any further information, but right now I have no idea what could be useful. (I am running JMRI under wayland if that matters).

Thanks,

dsand47, 2023-12-25:
The first step is to use Java 11.

Make sure that you have the latest changes to the master branch.  There have been several recent changes to the new serial library.

Make sure that the DCC++ USB connection is visible in the /dev directory and your user has the dialout group.

Dave Sand

voroskoi, 2023-12-25:
Hi Dave,

On Mon, Dec 25, 2023 at 7:20 PM Dave Sand ***@***.***> wrote:
>
> The first step is to use Java 11.

I have installed openjdk11, ran `ant clean` and `ant debug`, but
nothing changes.

> Make sure that you have the latest changes to the master branch. There have been several recent changes to the new serial library.

I have made a `git pull` just before building. I am on `a5484591ac`.

> Make sure that the DCC++ USB connection is visible in the /dev directory and your user has the dialout group.

```
jmri  master is 📦 v5.7.1-SNAPSHOT via ☕ v11.0.21 ❯ ls -la /dev/ttyACM0
crw-rw----    1 root     dialout   166,   0 Dec 25 20:03 /dev/ttyACM0
jmri  master is 📦 v5.7.1-SNAPSHOT via ☕ v11.0.21 ❯ id
uid=1000(voroskoi) gid=1000(voroskoi)
groups=10(wheel),18(audio),20(dialout),23(input),27(video),28(netdev),103(seat),1000(voroskoi)
```
It does not matter if my arduino is connected via usb or not, I get
the same error message.

Thanks,

dsand47, 2023-12-25:
I recommend that you do an "ant realclean" once.  I have never used "ant debug".  I use "ant panelpro".

You might try changing the DCC++ connection to the simulated mode.  This should confirm that JMRI is working.

Dave Sand

voroskoi, 2023-12-25:
Hi Dave,

On Mon, Dec 25, 2023 at 9:53 PM Dave Sand ***@***.***> wrote:
> I recommend that you do an "ant realclean" once. I have never used "ant debug". I use "ant panelpro".

Just tried `ant realclean`, but got the same results. I also start
with `ant panelpro`, but before that the JMRI build page suggests
running `ant debug` to create a debug build I guess.
It does not make any difference as I can tell.

> You might try changing the DCC++ connection to the simulated mode. This should confirm that JMRI is working.

As I said before, JMRI works fine, I can control my DCC-EX
CommandStation via WiThrottle server from EngineDriver app, no issues
with that.
However when I try to edit these tables it crashes instantly. I have
also tried with simulated DCC++, but crashes with that as well.

Thanks,

bobjacobsen, 2023-12-25:
Is it when you try to _edit_ the tables, or when you try to _open_ the tables?

The first time you open one of the tables, it loads some infrastructure for all of them.  That includes the audio libraries.  It’s a long shot, but if it’s crashing on opening the tables, something wrong in those audio tables might be the problem.

voroskoi, 2023-12-25:
Hi Bob,

On Mon, Dec 25, 2023 at 10:13 PM Bob Jacobsen ***@***.***> wrote:
> Is it when you try to _edit_ the tables, or when you try to _open_ the tables?

Sorry, when I _open_ them. So I have never seen the tables.

> The first time you open one of the tables, it loads some infrastructure for all of them. That includes the audio libraries. It’s a long shot, but if it’s crashing on opening the tables, something wrong in those audio tables might be the problem.

Are You talking about the files under `JMRI/resources/sound`? They
seems to be fine.

dsand47, 2023-12-25:
Currently you can start JMRI using DecoderPro or PanelPro, start the WiThrottle server on JMRI and run trains on the DCC_EX command station via the USB interface using the EngineDriver app.  Correct?

Dave Sand

voroskoi, 2023-12-25:
Hi,

On Mon, Dec 25, 2023 at 11:02 PM Dave Sand ***@***.***> wrote:
> Currently you can start JMRI using DecoderPro or PanelPro, start the WiThrottle server on JMRI and run trains on the DCC_EX command station via the USB interface using the EngineDriver app. Correct?

Yes, correct. I can even use DecoderPro for programming my train, all
works fine. I just can not open the signal/occupancy tables.

dsand47, 2023-12-25:
Post the contents of the "/home/voroskoi/.jmri/log/session.log" file after running a PanelPro session with WiThrottle working.

Dave Sand

bobjacobsen, 2023-12-25:
@klk32003 Do you have the recipe for installing the audio libraries on Linux?  I suspect the problems in this Issue may be due to them not being installed correctly. Is there a way to check?

voroskoi, 2023-12-26:
Hi,

On Mon, Dec 25, 2023 at 11:40 PM Dave Sand ***@***.***> wrote:
> Post the contents of the "/home/voroskoi/.jmri/log/session.log" file after running a PanelPro session with WiThrottle working.

Telling the truth I do not really understand why this is interesting,
but here is the log:
```
~/.jmri/log  master [!] ❯ cat session.log
2023-12-26T09:33:33,222 util.Log4JUtil                        INFO  -
****** JMRI log ******* [main]
2023-12-26T09:33:33,259 util.Log4JUtil                        INFO  -
This log is stored in file: /home/voroskoi/.jmri/log/session.log
[main]
2023-12-26T09:33:33,260 util.Log4JUtil                        INFO  -
This log is appended to file: /home/voroskoi/.jmri/log/messages.log
[main]
2023-12-26T09:33:33,321 apps.Apps                             INFO  -
PanelPro version 5.7.1plus+voroskoi+20231226T0832Z+Rd58dce8d6c starts
under Java 11.0.21 on Linux amd64 v6.6.8-0-lts at Tue Dec 26 09:33:33
CET 2023 [main]
2023-12-26T09:33:33,444 apps.Apps                             INFO  -
Starting with profile My_JMRI_Railroad.3f5cae03 [main]
2023-12-26T09:33:33,618 node.NodeIdentity                     INFO  -
Using 2bd8912f-f6c6-45c3-b549-a8eca3204f03 as the JMRI storage
identity for profile id 3f5cae03 [AWT-EventQueue-0]
2023-12-26T09:33:33,702 xml.AbstractSerialConnectionConfigXml INFO  -
Starting to connect for "DCC++" [main]
2023-12-26T09:33:33,830 serial.DCCppAdapter                   INFO  -
Connecting DCC++ to ttyACM0 Serial Device (acm) [main]
2023-12-26T09:33:33,835 serial.DCCppAdapter                   INFO  -
Port ttyACM0 Serial Device (acm) opened at 115200 baud, sees DTR: true
RTS: true DSR: false CTS: true DCD: false flow: NONE [main]
2023-12-26T09:33:33,981 dccpp.DCCppInitializationManager      INFO  -
DCC++ Initialization Complete [main]
2023-12-26T09:33:35,830 util.FileUtilSupport                  INFO  -
File path program: is /home/voroskoi/code/jmri/ [main]
2023-12-26T09:33:35,830 util.FileUtilSupport                  INFO  -
File path preference: is /home/voroskoi/.jmri/My_JMRI_Railroad.jmri/
[main]
2023-12-26T09:33:35,830 util.FileUtilSupport                  INFO  -
File path profile: is /home/voroskoi/.jmri/My_JMRI_Railroad.jmri/
[main]
2023-12-26T09:33:35,831 util.FileUtilSupport                  INFO  -
File path settings: is /home/voroskoi/.jmri/ [main]
2023-12-26T09:33:35,831 util.FileUtilSupport                  INFO  -
File path home: is /home/voroskoi/ [main]
2023-12-26T09:33:35,831 util.FileUtilSupport                  INFO  -
File path scripts: is /home/voroskoi/code/jmri/jython/ [main]
2023-12-26T09:33:36,438 dccpp.DCCppCommandStation             INFO  -
Station Type set to 'DCC-EX' [AWT-EventQueue-0]
2023-12-26T09:33:36,439 dccpp.DCCppCommandStation             INFO  -
Build set to '3bddf4d' [AWT-EventQueue-0]
2023-12-26T09:33:36,439 dccpp.DCCppCommandStation             INFO  -
Version set to '5.0.7' [AWT-EventQueue-0]
2023-12-26T09:33:36,444 dccpp.DCCppCommandStation             INFO  -
changing maxNumSlots from 12 to 50 [AWT-EventQueue-0]
2023-12-26T09:33:44,830 PanelPro.PanelPro                     INFO  -
Main initialization done [main]
2023-12-26T09:34:06,866 withrottle.FacelessServer             INFO  -
Published ZeroConf service for 'my jmri
railroad._withrottle._tcp.local.' on
fe80:0:0:0:ca09:a8ff:fe54:e7f0%wlan0:12090 [WiThrottleServer]
2023-12-26T09:34:06,870 withrottle.FacelessServer             INFO  -
Published ZeroConf service for 'my jmri
railroad._withrottle._tcp.local.' on
fe80:0:0:0:ef5:4227:e16e:cd0f%tailscale0:12090 [WiThrottleServer]
2023-12-26T09:34:06,871 withrottle.FacelessServer             INFO  -
Published ZeroConf service for 'my jmri
railroad._withrottle._tcp.local.' on 100.89.208.113:12090
[... 94 lines cut ...]
2023-12-26T09:35:32,154 withrottle.DeviceServer               ERROR -
readLine failure limit exceeded, ending thread run loop for device
'Andriska' [DeviceServer-1]
2023-12-26T09:36:01,503 withrottle.FacelessServer             INFO  -
Creating new WiThrottle DeviceServer(socket) on port 12090, waiting
for incoming connection... [WiThrottleServer]
2023-12-26T09:36:01,506 withrottle.DeviceServer               INFO  -
Received Name: Andriska [DeviceServer-2]
2023-12-26T09:36:15,218 withrottle.DeviceServer               INFO  -
Received Name: Andriska [DeviceServer-2]
2023-12-26T09:36:23,307 withrottle.DeviceServer               INFO  -
Received Name: Andriska [DeviceServer-2]
2023-12-26T09:36:31,295 withrottle.DeviceServer               INFO  -
Received Name: Andriska [DeviceServer-2]
2023-12-26T09:36:32,202 withrottle.DeviceServer               WARN  -
readLine from device 'Andriska' failed, consecutive error # 1
[DeviceServer-2]
2023-12-26T09:36:36,233 managers.DefaultShutDownManager       INFO  -
Normal termination complete [Thread-32]
```

As You can see I have closed EngineDriver for the first time, then
connected again. All worked fine, but my original problem remains.

Thanks,

klk32003, 2023-12-26:
> The first time you open one of the tables, it loads some infrastructure for all of them.  That includes the audio libraries.  It’s a long shot, but if it’s crashing on opening the tables, something wrong in those audio tables might be the problem.

Right, ```Tools->Tables``` initialises jmri.Audio.
The success message is something like ```audio.JoalAudioFactory  INFO  - Initialised JOAL using OpenAL...```
If it fails the message would be ```jmri.jmrit.audio.JoalAudioFactory  WARN  - Error initialising JOAL```

> Do you have the recipe for installing the audio libraries on Linux? I suspect the problems in this Issue may be due to them not being installed correctly. Is there a way to check?

```jmri.Audio``` requires OpenAL installed.
A quick check is ```$ dpkg --get-selections | grep "openal*"```

The install is done by ```$ sudo apt-get install libopenal-dev```
See https://www.jmri.org/help/en/html/tools/Audio.shtml#playing for details.

dsand47, 2023-12-26:
The second log is interesting because the JMRI environment changed.

The original posting had a serial connection error, but you indicated that WiThrottle worked OK.  This means the serial connection is actually working.

Here is a snippet from the original posting.

> [java] 18:42:16,320 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3ee154a2 [main]
>      [java] 18:42:16,388 jmri.util.node.NodeIdentity           INFO  - Using [...]

Here are the same lines from the current posting.

> 2023-12-26T09:33:33,444 apps.Apps INFO - Starting with profile My_JMRI_Railroad.3f5cae03 [main]
> 2023-12-26T09:33:33,618 node.NodeIdentity INFO - Using 2bd8912f-f6c6-45c3-b549-a8eca3204f03 as the [...]

The profile id and the node identity changed.  The serial connection is now working.  

This means that whatever was done between the two sessions resolves the errors in the first session log.

Now that you have a clean PanelPro session you can try adding a sensor.  Just to keep things clean, I recommend that you do not start WiThrottle.

Dave Sand

dsand47, 2023-12-26:
Since you are building JMRI, you could try removing the audio table from the table initialization process.
```
diff --git a/java/src/jmri/jmrit/beantable/ListedTableFrame.java b/java/src/jmri/jmrit/beantable/ListedTableFrame.java
index c010654528..a16f7c1d0b 100644
--- a/java/src/jmri/jmrit/beantable/ListedTableFrame.java
+++ b/java/src/jmri/jmrit/beantable/ListedTableFrame.java
@@ -115,7 +115,7 @@ public class ListedTableFrame<E extends NamedBean> extends BeanTableFrame<E> {
             } // requires restart after changing the interface setting (on Display tab)
             addTable("jmri.jmrit.beantable.SectionTableAction", Bundle.getMessage("MenuItemSectionTable"), true);
             addTable("jmri.jmrit.beantable.TransitTableAction", Bundle.getMessage("MenuItemTransitTable"), true);
-            addTable("jmri.jmrit.beantable.AudioTableAction", Bundle.getMessage("MenuItemAudioTable"), false);
+//             addTable("jmri.jmrit.beantable.AudioTableAction", Bundle.getMessage("MenuItemAudioTable"), false);
             addTable("jmri.jmrit.beantable.IdTagTableTabAction", Bundle.getMessage("MenuItemIdTagTable"), false);
             addTable("jmri.jmrit.beantable.RailComTableAction", Bundle.getMessage("MenuItemRailComTable"), true);
             ListedTableFrame.setInit(true);
```

This might be the cause of the crashes.

voroskoi, 2023-12-26:
Hi Dave,

On Tue, Dec 26, 2023 at 5:47 PM Dave Sand ***@***.***> wrote:
> The second log is interesting because the JMRI environment changed.
>
> The original posting had a serial connection error, but you indicated that WiThrottle worked OK. This means the serial connection [...]

Oh, I terribly sorry, I did not meant to mislead You. I just run JRMI
for the first time without the board plugged in, that is the reason
for the error and when You suggested I tried it with the board plugged
in too, but did not make any difference, so I did not update my logs.
The change of the ID is because I also did `rm -r ~/.jmri` when I
rebuilt it with java11 to make sure everything is cleaned up.
Again, I am sorry for not being clear on that.

And here is the good news: openal was not installed on my machine, but
I have installed it, rebuilt JMRI, did not help. However Your
suggested patch does the trick, so You were right, it is something
related to sound.

Btw, I have installed openal-soft-dev (I am on Alpine Linux) and
rebuilt PanelPro and I need the patch anyway, but it is a step forward
for sure, thank You!

dsand47, 2023-12-26:
I took a brief look at Alpine Linux.  It is quite different than the typical Linux distributions.

I suspect that it does not include the necessary components to support the sound libraries used by JMRI/Java.

klk32003, 2023-12-26:
You can create a logfile. Execute the following lines before you run PanelPro:
```
export ALSOFT_LOGLEVEL=3
export ALSOFT_LOGFILE=alsoftlog.txt
```

The logfile can be found in ```/home/voroskoi/code/jmri/```.

klk32003, 2023-12-27:
@voroskoi
This quick test might be worth a try:
* navigate to the lib/linux folder of your source code,
* choose the appropiate subfolder (x86_64?),
* remove the file libopenal.so and store it to a save place.

JMRI will now take your installed ALSoft version.

voroskoi, 2023-12-29:
Hi,

OpenAL logging does not work, I do not get any logs.

On Wed, Dec 27, 2023 at 3:21 PM klk32003 ***@***.***> wrote:
> @voroskoi
> This quick test might be worth a try:
>
> navigate to the lib/linux folder of your source code,
> choose the appropiate subfolder (x86_64?),
> remove the file libopenal.so and store it to a save place.
>
> JMRI will now take your installed ALSoft version.

This indeed work, it also fixes the openal initialization problem, so
with this openal works fine too.

I assume it is because Alpine uses `musl` libc instead of `glibc`, so
those prebuilt objects will not work. Why do You ship those instead of
relying on distribution packages?

Thanks,

klk32003, 2023-12-30:
Hi,

I believe that Dave was right when he said that Alpine Linux (musl based) does not include the necessary components to support the sound libraries. I think an audio backend was missing and with your latest installation step those sound libraries were included.
Of course, I could also be wrong. If I am right you could use the libopenal.so again. At least it should be possible to open tables.

JMRI is using [JOAL](https://jogamp.org/joal/www/). The latest release contains OpenAL-Soft. But that might not answer your last question. Maybe there was no package solution available with the Java bindings feature plus an audio backend.

closed 2024-03-25 by github-actions[bot]