JMRI/JMRI #12767: Rename sample jython/MQTT/MqttSignalHead.py to MqttSignalMast.py

ggee, 2023-12-26:
To better reflect the function, I think it should be called MqttSignalMast.py.  I made the below new MqttSignalHead.py that actually publishes the Signal Head to topics.

```
import java
import java.beans
import jmri
import apps

# 0  = Dark
# 1  = Red
# 2  = Flashing Red
# 4  = Yellow
# 8  = Flashing Yellow
# 16 = Green
# 32 = Flashing Green

mqttAdapter = jmri.InstanceManager.getDefault( jmri.jmrix.mqtt.MqttSystemConnectionMemo ).getMqttAdapter()
mqttSignalTopic = mqttAdapter.getOptionState("14")

class SignalHeadChanger( java.beans.PropertyChangeListener ):

  def propertyChange( self, event ):
    if (event.propertyName == "Appearance") :
      signalHead = event.source
      mqttAdapter.publish(mqttSignalTopic + signalHead.getSystemName(), signalHead.getAppearanceName())

listener = SignalHeadChanger( )
for signal in signals.getNamedBeanSet():
  signal.addPropertyChangeListener(listener)
```

danielb987, 2023-12-26:
@ggee 
Thank you and welcome to the JMRI project!

@bobjacobsen 
How do we handle this? Can we just do the rename and add the new script, or is it a risk that we break something for users if we do that?

dsand47, 2023-12-26:
The existing MqttSignalHead.py script is using signal heads.  Changing the name only confuses things since it is definitely not referring to masts.

dsand47, 2023-12-26:
The only change that I see is replacing a static list of heads with a getNamedBeanSet.

ggee, 2023-12-26:
Sorry, you are write.  For some reason I thought it was MAST.  I must have been looking at some other file.

ggee, 2023-12-26:
> The only change that I see is replacing a static list of heads with a getNamedBeanSet.

Yah, I guess that is the only difference.  My script gets all heads while your sample uses a static list supplied.

And trying to get the mqtt channel name from preferences was tricky.

bobjacobsen, 2023-12-27:
Sorry to come to this late.  I think the script at the top is an improvement because it doesn't require manual configuration, and it uses the topic string from the connection preferences. We don't have native SignalHead support, as that code is really really hard to work with, so having the script is useful.  But I don't think it should be renamed, and we do have native MQTT support for signal heads.

danielb987, 2023-12-29:
I have added this script in PR #12775. I wasn't sure if I should replace the old script or add this as a new script, so I added this as a new script.

danielb987, 2023-12-30:
@ggee 
PR #12775 is now merged and your script will be included in the next JMRI test version 5.7.2 as `jython/MQTT/MqttSignalHead2.py`.

ggee, 2024-01-01:
Great.

closed 2024-03-25 by github-actions[bot]