JMRI/JMRI #12782: Feature Request: Add peakCurrent for decoders

tazzytazzy, 2023-12-30:
A lot of decoders have a "continuous current" and a brief "peak current". JMRI doesn't differentiate between these two. It appears that with JMRI, 'maxCurrent' is really 'continuous current'.

Any thoughts on adding 'peakCurrent' and _possibly_ changing 'maxCurrent' to something like 'maxContinuousCurrent'.  I don't want to confuse people with peak vs max.

bobjacobsen, 2023-12-30:
Right now, there are about 2100 instances of "maxMotorCurrent" and 1200 "maxTotalCurrent" values.  

There are about 3400 separate models defined, so we don't have all that large a coverage for these attributes. And I'm not sure whether people have entered peak or average numbers in all those.

Would we need separate peak values for motor current and total current?

I'm a bit hesitant to change the existing names.  I'm not sure what or who might be depending on those.  But if we have new attributes with "peak" in their name, perhaps that would make it clear which was which.

You can see the rest of the attributes for the model element [here](https://www.jmri.org/xml/schema/decoder-4-15-2.xsd#id13). We can add comments to the existing value to make their meaning clearer, but I'm not sure how many people ever encounter that reference.

tazzytazzy, 2023-12-31:
Good point about changing the attribute names, let's not do that. The UI can differentiate that when being displayed as long as it's noted in the xsd file as well.

In the xsd file, there's currently:
- maxMotorCurrent - In amps, the maximum continuous motor power.
- maxTotalCurrent - In amps, the maximum continuous power available to motor and functions.
-- Annotations are missing and should be added to the xsd, which I can do.

I would like to suggest these new attributes:
- peakMotorCurrent- In amps, the maximum total short-term peak motor power. Also known as 'stall current peak', this is the peak current the decoder can handle without blowing one of its motor transistors up.
- peakTotalCurrent - The absolute max power that the decoder can handle with functions and motor in use without the decoder being damaged.

Optionally, an idleCurrent - The amount of current used with the decoder sitting on the track, but no functions being on (lights, sounds, animations, etc). This could help bring awareness about parasitic power draw of a layout has a bunch of decoders on a layout, but not being used.

Existing entry:
```xml
<model model="DG383" numOuts="8" numFns="14" maxMotorCurrent="3A (peak=5A)" formFactor="O/G" connector="9pin">
```

Updated entry, values added were made up for example:
```xml
<model model="DG383" numOuts="8" numFns="14" maxMotorCurrent="3A" peakMotorCurrent="5A" peakTotalCurrent="6A" formFactor="O/G" connector="9pin">
```

Could you help me to determine how to uniquely identify a decoder? Is it a combination of high/low id's along with the modelname/productID? There's a productID entry, but there are currently 1068 duplicate productID's used. There are also 379 duplicate 'model' entries. What do you think about adding a unique ID to each model? Something like up to 4 letters for MFG and then whatever random/sequential number for the model.

And lastly, what about adding a sub-model? For example, there's currently a listing for DG583, but there's no actual product sold as that, it's either DG583S or DG583AR, but basically the exact same thing with different plugs. This would allow people to select their specific variant without having to define another model.

Sub-model idea:

```xml
<model model="DG583" numOuts="8" numFns="14" maxMotorCurrent="5A (peak=10A)" formFactor="O/G" connector="9pin" uniqID="DGTX410">
  <subModel model="DG583S" uniqID="DGTX410.1" />
  <subModel model="DG583AR" uniqID="DGTX410.2" />
  <output name="4" label="Violet" connection="wire"/>
  <output name="5" label="Brown" connection="wire"/>
  <output name="6" label="Wht/Yel" connection="wire"/>
  <output name="7" label="Wht/Grn" connection="wire"/>
  <output name="8" label="Wht/Blu" connection="wire"/>
  <size length="2.23" width="1.45" height="0" units="inches"/>
 </model>
```

I just pulled these statistics from the XML Files.

- model: 3433, Unique: 3054
- numOuts: 3168, Unique: 19
- numFns: 2976, Unique: 30
- productID: 2525, Unique: 1457
- maxMotorCurrent: 2098, Unique: 116
- connector: 1690, Unique: 19
- formFactor: 1584, Unique: 39
- lowVersionID: 1481, Unique: 87
- comment: 1301, Unique: 1073
- highVersionID: 1276, Unique: 78
- maxTotalCurrent: 1231, Unique: 78
- maxInputVolts: 991, Unique: 24
- replacementModel: 643, Unique: 418
- replacementFamily: 640, Unique: 38
- show: 401, Unique: 2
- extFnsMRC: 103, Unique: 2
- extFnsESU: 99, Unique: 2
- nmraWarrant: 47, Unique: 2
- maxFnNum: 39, Unique: 3
- nmraWarrantStart: 37, Unique: 11
- extFnsNmraF13: 17, Unique: 2
- extFnsNmraBSC: 2, Unique: 1
- developerID: 2, Unique: 2
- options: 1, Unique: 1
- nmraWarrantEnd: 1, Unique: 1

Why I'm doing this: https://dccwiki.com/JMRI:Main_Page 
It's only 2 days old, so I'm still working on the layout, formatting, and display tools.

bobjacobsen, 2023-12-31:
We should probably move this to the JMRI developers list, as it’s becoming a discussion that others may have contributions to.

https://jmri-developers.groups.io/g/jmri

Are you a member there?

Would it be OK if I reposted your last message and replied there?

tazzytazzy, 2023-12-31:
I just applied for membership to that group. Yes, I'm fine moving it there.

bobjacobsen, 2024-01-23:
The decoder-4-15-2.xsd Schema definition was updated to include this.

closed 2024-01-23 by bobjacobsen