JMRI/JMRI #12794: Web panel ignoring element "level"

mstevetodd, 2024-01-01:
When adding "shapes" to a panel, the user can specify a "level" for the shape, to control overlapping "shapes".
This level is ignored by the web panel.
This issue is to correct this, most likely by sorting the "shapes" on the server side before adding them to the returned xml. This sorting will allow the client to continue to draw efficiently as each element is processed in the xml.
Example config attached, NY&HV_RR_Carfloat_Configs panel shows the problem clearly where the gray rectangles should be on top of the yellow rounded rectangle.
[DR5000.zip](https://github.com/JMRI/JMRI/files/13802471/DR5000.zip)

mstevetodd, 2024-01-01:
This was found and discussed in #12663

bobjacobsen, 2025-01-27:
See also "Web server does not recognize shape POLYGON" #13811

closed 2026-01-12 by github-actions[bot]

mvanhorn, 2026-06-14:
Submitted a fix in https://github.com/JMRI/JMRI/pull/15216.