JMRI/JMRI #12804: Can't unselect Consists from Preferences/Defaults

jerryg2003, 2024-01-03:
When Preferences/Defaults is first displayed, no selection is made for Consists.  [If you look in profile.properties, you find jmri-managers.defaults.jmri-ConsistManager=Internal even though "Internal" does not have a selectable button for Consists.

When you select Consists for some connection that has Consists (e.g. LocoNet or MQTT), you can't un-select unless another connection type has an active Consists button (in which case, you can switch to that).

Seems like you should always be able to unselect any default.

danielb987, 2024-01-03:
What's the purpose to unselect consists? Why do you want to do that?

jerryg2003, 2024-01-04:
Selecting or unselecting Consist should presumably say whether JMRI can send or receive messages to a consist controlled via an MQTT-aware train controller.  If you can't unselect it, then you lose the ability to restrict these messages.  And if you can't unselect it, they why have it as an option?  [Presumably the author, @DeanCording has an answer to this. I'm just pointing out the inconsistency.]

Further testing:  it appears to make no difference whether Consist is selected or not:  even with it unselected (which I did manually by editing the profile.properties file to the initial state which is jmri-managers.defaults.jmri-ConsistManager=Internal), JMRI still subscribed and sent messages to the consist when a throttle was opened and set to a consist number.

DeanCording, 2024-01-09:
I'll have a look at in conjunction with #12811 .  Thanks for raising it.

DeanCording, 2024-01-15:
One of the lessons you learn as a JMRI developer is not to stray too far outside of the module you are working on because that way lies madness.  This is one of the cases why.....

From the point of view of the different connection throttle modules, they don't care about what the config default settings are - they just get told that they are controlling a particular loco or consist address, and they just do what they're told. 

- Consists aren't tied to particular connections - you can control any consist from any throttle connected to any connection.
- If you open a Throttle from the Consist Tool, it opens the default throttle set in the Preferences/Defaults/Throttles (but it doesn't indicate which connection the throttle is using)

What the Preferences/Default/Consist setting does is specify which connection is used to program the consist.  So if you have an more than one type of connection, whichever one is specified as the default for Consists will be used to configure the Consist, irrespective as to which one is used as the throttle, even if you select the consist from the Throttle.  This is definitely a bug for Command Station Consists because the throttle connection needs to be the one controlling the consist.

In the case of having an MQTT and a LocoNet connection, if you set Preferences/Default/Consists to LocoNet, you will notice that MQTT doesn't get sent the consist configuration when you use the MQTT throttle.  Whereas, if you set Preferences/Default/Consists to MQTT, MQTT will be sent the consist configuration when you use the MQTT throttle (but not if you use the LocoNet throttle for reasons - the different ways that consists are configured lead to inconsistencies when mixing connections).

As for the original issue about not being able to set the default for Consists (or any other option) back to Internal, herein is where the true madness lies...

A fresh JMRI config has no connections.  However, in the Preference/Defaults window it shows an Internal connection, which can't be selected because it is not real.  However, if you create a new Connection with the System Manufacturer of None, you will get a real internal connection which you can select for default connections.  This real internal connection will replace the original not-real internal connection BUT only if the real Internal connection is the first connection.  If it isn't the first connection then you get both the real and the not-real internal connections - the real one which you can select and the not-real one that you can't.

[image]

[image]

So, the issue is that JRMI shows an Internal connection in the Preference/Defaults that is not real when the first real connection is not Internal.  It really shouldn't show anything when there are no connections because, well, there are no connections.  If you want an Internal connection, then you can just create a real one which works as advertised.

This should be a simple fix but as it isn't an area of the code I'm familiar with, its not a rabbit hole I'm inclined to go down.

closed 2024-04-08 by github-actions[bot]