JMRI/JMRI #12807: LogixNG throwing NullPointerException when using Modules

jbnx42, 2024-01-03:
See the original discussion at the JMRI User Group [Topic 103463824](https://groups.io/g/jmriusers/topic/103463824). The files I used are currently in the ProblemsBeingWorkedOn folder there. I wanted to enter the issue here so it doesn't get lost. It might well be that I made a blunder, but if I managed it, other users might do the same. A quick look at the logs and the source code makes me think this is a genuine bug. 

When I created an over-complicated LogixNG using Modules, I managed to get a Null Pointer Exception. 

Both logged errors occurred in jmri.jmrit.logixng.implementation.DefaultMaleDigitalExpressionSocket.evaluate.
One is at line 60: conditionalNG.getSymbolTable().createSymbols(_localVariables);
and the other is at line 75: conditionalNG.getSymbolTable().removeSymbols(_localVariables);

If you need any more information, I am happy to help!

Thanks,
John B. on the Delmarva Peninsula.

danielb987, 2024-01-03:
Thanks. You are right that it's there the exception happens. But that doesn't tell you why it happens. The cause is probably that `conditionalNG` is null. But it shouldn't be null here. So something else is happening and I don't yet know what that is.

danielb987, 2024-01-18:
@jbnx42 
I'm unable to reproduce the error. Which steps should I do to trigger the error?

jbnx42, 2024-01-18:
Hi Daniel,

And I owe you an apology. You asked in the user's group about this a week ago; I did not see that message. Sorry about that.

It has been a couple of weeks.  I certainly can get the error if I choose Evaluate when Editing. I don't remember if I saw it otherwise. Here are the steps:

Open PanelPro. Mine is 5.4 under Java 11.0.21 under Windows 10
Show the system console so you can see when the error occurs.
Load Panel (File -> Load File Content and panels... Choose the file I provided)
Tools -> Tables -> LogixNG -> LogixNGs
select "LH583 Show Aspect" and choose Edit
select "LH583 Show Stop" (top line) and click Edit
in the Editor Dialog, right-click on the second line, "If Digital Formula: E1 && !E2," and choose Evaluate.

the result is a dialog box "The result of the expression is: true" [OK]
However, the System Console also shows the error. [NOT OK]

Thanks,
John B.

danielb987, 2024-01-18:
Thank you!

It was easy to fix once I realize where the problem was.
https://github.com/JMRI/JMRI/pull/12842/files?diff=unified&w=1

PR #12842 (2024-01-20): Fix bug in LogixNG
Fixes #12807.
closed 2024-01-20 by danielb987