JMRI/JMRI #12808: Ellipse & Circle Shapes not recognized by web server

StCircumstance, 2024-01-03:
See attached labeled photos of a control panel. The zip file I uploaded last week has the circle on panel named ....Overview.

[image]
[image]
[image]

mstevetodd, 2024-01-04:
@StCircumstance 
The example panel you sent earlier only shows a circle, but I figured out how to edit the Control Panel and add an ellipse.
I've added code to draw both shapes and am testing it now with the various permutations.
One issue is all drawn objects are drawn at level=2, so that will still be an issue until #12794 is picked up and completed. I had to lower your images to level 1 to get the drawing to show up.
I'll let you know when a test version is available for you to try.

StCircumstance, 2024-01-04:
That's correct, that export only showed the circle, not the ellipse.
Awesome, I can get around those minor complications, thank you Steve!

mstevetodd, 2024-01-04:
@StCircumstance I have submitted the PR which contains the fix. It'll be in a development release in a day or so, then test and production releases later.
If you'd like to try it before then, you can download the panel.js here: http://jmri.mstevetodd.com:12080/dist/web/js/panel.js
and replace the one in your web/js folder. Note: this link is temporary, as it is my demo server, where I update JMRI quite often.

Important, this does not fix the level issue, so all drawn objects are still locked at level 2. Move anything that should be "under" drawn objects to level 1.

StCircumstance, 2024-01-04:
great thank you Steve!

closed 2024-01-06 by danielb987

mstevetodd, 2025-02-11:
Sorry, I didn't mean to indicate that I was working on this. It may be quite a while before I have enough time to do so. 
Hopefully, someone else will pick up this issue and take care of it for you.

StCircumstance, 2025-02-11:
Ok no problem I just wanted to get an idea.

PR #12810 (2024-01-06): web panel.js handle positionableCircle and positionableEllipse
closes #12808