JMRI/JMRI #12811: MQTT subscription/unsubscription for new receive topics

jerryg2003, 2024-01-04:
New (as of 5.5.4) MQTT topics include Send/Receive for Power, Throttle (speed), Direction, Function, Consist.  MQTT SUBSCRIBE and UNSUBSCRIBE (for the Receive topics for each) appear to be inconsistent: with the way JMRI handles other topics:

JMRI never does an UNSUBSCRIBE for topics such as Sensor Receive, Reporter Receive, etc.,, either when an MQTT object is deleted or when JMRI is closed -- which appears to work just fine.
---
Throttle, Direction, Function Receive: correctly SUBSCRIBE when a throttle is opened and assigned to a particular loco or consist (regardless of whether Consist is selected in Preferences/Defaults, see Issue #12804 ) , Throttle and Direction are UNSUBSCRIBEd  when the throttle is closed (correctly, I would say), and also when JMRI is closed (inconsistent with other JMRI MQTT Receive topics, but likely OK).  Function is never UNSUBSCRIBEd in either case (why or why not?).

I do not know if there are any operational implications for these inconsistencies in UNSUBSCRIBE, but they are inconsistent with the way JMRI handles other topics.  
---
Power Receive: SUBSCRIBEd even if MQTT Power not checked as Preferences/Default; never UNSUBSCRIBED.

I do not know how JMRI will react in receiving a Power command via MQTT when Power Control is not selected for MQTT but likely it shouldn't be SUBSCRIBED in this case.  [@DeanCording - what did you have in mind here?]
---
Which is the proper behavior and shouldn't all MQTT topics behave similarly as regards SUBSCRIBE and UNSUBSCRIBE?  As few people are yet using these new MQTT capabilities, it is a good time to see if there are any issue here.

DeanCording, 2024-01-09:
Thanks for reporting this.  I'm going to need a little time to look through the logic of the various components to check the consistency for handling connections.

DeanCording, 2024-01-16:
> Function is never UNSUBSCRIBEd

That was a bug - fixed in the PR

> Power Receive: SUBSCRIBEd even if MQTT Power not checked as Preferences/Default; 

JRMI is designed on the basis that if a connection has a known feature, such as power control, then that connection has that feature whether it actually works or not. The MQTT connection is somewhat unique in that it doesn't connect to a specific piece of hardware with known features.  JRMI is happy to shout out commands irrespective of if anyone is listening; that's why it's best to have separate command and status topics.  

The Preference/Default/Power Control just selects which connection is used by default for some controls, such as in the Throttle tool.   It doesn't enable/disable the feature for a connection.

> Power Receive never UNSUBSCRIBED.

> JMRI never does an UNSUBSCRIBE for topics such as Sensor Receive, Reporter Receive, etc.,, either when an MQTT object is deleted or when JMRI is closed

I fixed a bug that didn't remove the subscription when you deleted a Sensor, Reporter, etc.

However, JRMI really doesn't handle shutting down stateful connections well.   For Throttles, it goes through the process of disposing of each Throttle on shutdown, just so your trains don't keep running out of control.  For everything else, it just seems to walk away without turning out the lights.

> when you select MQTT as Default for "Power Control" or "Throttle," it takes away the ability to control these through any other active connections you might have. 

No, that isn't how the Preferences/Default works.  The connection specified in the Preferences/Default defines the default connection used for some tools.  For example, when you open a Throttle from the Consist tool, it uses the default throttle connection.  The Power button on the Throttle tool uses the default Power connection.

However, you can select a Throttle for a specific connection through the Tools/Throttles/Throttle for Connection menu item.  Likewise, opening the Power Control tool from the Tools menu allows you to control individual power connections.

bobjacobsen, 2024-03-04:
Has this been fixed? PR #12841 seems to address it.

DeanCording, 2024-03-04:
I haven't heard anything more from @jerryg2003 but I believe I've address all of the issues.

closed 2024-03-04 by bobjacobsen