JMRI/JMRI #12814: LogixNG Request for sensor

sidlo64, 2024-01-08:
LogicNG has action Request update of all sensors.
Unfortunately it doesn't work with XpressNet.
New function Request for sensor would be appropriate.

1) The LogixNG action should call the method DigitalIO.requestUpdateFromLayout()
   This means that it is available for every named bean that implements DigitalIO, for example turnout and sensor.

2) Not all connections implement requestUpdateFromLayout(). LocoNet sensor does not.
   The LogixNG action needs to tell that it doesn't work for every connection.

3) The "Request Update All Sensors" should also be updated to tell that it doesn't work for every connection.

https://jmri-developers.groups.io/g/jmri/topic/what_new_features_do_you_want/87367048?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,87367048,previd%3D1704727240814061839,nextid%3D1703442380621021266&previd=1704727240814061839&nextid=1703442380621021266

danielb987, 2024-01-13:
The action `Request update of sensor` is added in PR #12832.

danielb987, 2024-02-08:
Closing this as completed.

closed 2024-02-08 by danielb987