JMRI/JMRI #12820: Automatic adjustment of the column width of the CV table in DecoderPro

uli-2016, 2024-01-10:
Is there a way to improve the column width and the window width for the CV table in DecoderPro? If there are indexed CVs, the column with the CV number and the window with the table are too small for a complete display. Individual columns can be widened, but the overall width cannot be changed, although there would be enough space on the sides.

[image]

bobjacobsen, 2024-01-10:
I can set the width to fill the pane, but then it looks a little weird.

[image]

Is that actually better?

uli-2016, 2024-01-10:
For me it looks better. I think this is the easiest solution.

closed 2024-01-11 by bobjacobsen