JMRI/JMRI #12835: Entry Exit - unrecoverable user mistake

sidlo64, 2024-01-15:
A beginner user builds a simple layout and generates Entry Exit pairs. The user does not know that he cannot not use Distant signal mast. Distant signal mast does not have the aspect Stop and cannot protect block. Incorrectly uses the signal mast sequence Exit - Distant - Home.

[image]

Builds a path from WEST to EAST. An exception occurs when Block 3 is occupied.

`15:47:29,239 jmri.implementation.VirtualSignalMast WARN  - attempting to set invalid aspect: null on mast: Distant [AWT-EventQueue-0] 
15:47:29,241 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: attempting to set invalid aspect: null on mast: Distant
    at jmri.implementation.VirtualSignalMast.setAspect(VirtualSignalMast.java:73) ~[jmri.jar:?]
    at jmri.implementation.DefaultSignalMastLogic.lambda$setMastAppearance$4(DefaultSignalMastLogic.java:1073) ~[jmri.jar:?]
    at jmri.util.ThreadingUtil.runOnGUI(ThreadingUtil.java:134) ~[jmri.jar:?]
    at jmri.util.ThreadingUtil.runOnLayout(ThreadingUtil.java:42) ~[jmri.jar:?]
    at jmri.implementation.DefaultSignalMastLogic.setMastAppearance(DefaultSignalMastLogic.java:1072) ~[jmri.jar:?]
    at jmri.implementation.DefaultSignalMastLogic.lambda$setSignalAppearance$2(DefaultSignalMastLogic.java:1047) ~[jmri.jar:?]
    at jmri.util.ThreadingUtil.lambda$runOnGUIDelayed$2(ThreadingUtil.java:290) ~[jmri.jar:?]`

The user will try to fix it. An exception occurs when you try to delete the Entry Exit pair Distant - Home. 

`15:48:13,589 jmri.implementation.VirtualSignalMast WARN  - attempting to set invalid aspect: null on mast: Distant [AWT-EventQueue-0]
15:48:13,590 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: attempting to set invalid aspect: null on mast: Distant
    at jmri.implementation.VirtualSignalMast.setAspect(VirtualSignalMast.java:73) ~[jmri.jar:?]
    at jmri.jmrit.entryexit.DestinationPoints.cancelClearInterlock(DestinationPoints.java:823) ~[jmri.jar:?]
    at jmri.jmrit.entryexit.DestinationPoints.dispose(DestinationPoints.java:1278) ~[jmri.jar:?]
    at jmri.jmrit.entryexit.Source.removeDestination(Source.java:173) ~[jmri.jar:?]
    at jmri.jmrit.entryexit.EntryExitPairs.deleteNxPairs(EntryExitPairs.java:927) ~[jmri.jar:?]
    at jmri.jmrit.entryexit.EntryExitPairs.deleteNxPair(EntryExitPairs.java:791) ~[jmri.jar:?]
    at jmri.jmrit.entryexit.AddEntryExitPairPanel$TableModel.deleteEntryExit(AddEntryExitPairPanel.java:493) ~[jmri.jar:?]
    at jmri.jmrit.entryexit.AddEntryExitPairPanel$TableModel.setValueAt(AddEntryExitPairPanel.java:445) ~[jmri.jar:?]
    at javax.swing.JTable.setValueAt(JTable.java:2730) ~[?:?]
    at javax.swing.JTable.editingStopped(JTable.java:4728) ~[?:?]`

There is no easy way to recover from this error.

[EntryExitDistant.jmri.zip](https://github.com/JMRI/JMRI/files/13941331/EntryExitDistant.jmri.zip)

dsand47, 2024-01-15:
I suspect that the problem is with the "appearance-entry_distant.xml" file.  It does not have a "specificappearances" section with a "danger" element.  Entry/Exit requires the danger element for any signal mast types assigned to signal masts within the Entry/Exit scope.

I noticed that there is no SML.  Except for "Turnouts only" NX pairs, SML is required.  If non exists, temporary SML will be created.  

EntryExitDistant.png (view on web) <https://github.com/JMRI/JMRI/assets/20986698/b270d194-baad-42df-9e04-394f0985ff6d>

sidlo64, 2024-01-16:
The distant signal mast does not have a dangerous aspect https://en.wikipedia.org/wiki/Application_of_railway_signals#Distant_signal
Yes, I know that Entry/Exit needs a danger element.
I can provide a replacement solution with minimal cost that works like a prototype. I don't need advice.
I intentionally made a sample example with a user mistake.
If a beginner makes this mistake, he gets into an unsolvable situation.