JMRI/JMRI #12850: Message.log not moving to message.1.log when full

PUlvestad, 2024-01-21:
Message.log not moving to message.1.log when full. It appears to just be deleted and starts over.

bobjacobsen, 2024-01-21:
Problem seems to be associated with this error message:
```
ERROR StatusConsoleListener Unable to copy file C:\Users\User\JMRI\log\messages.log to C:\Program Files (x86)\JMRI\${jmri.log.path}messages.1.log: java.nio.file.AccessDeniedException C:\Users\User\JMRI\log\messages.log -> C:\Program Files (x86)\JMRI\${jmri.log.path}messages.1.log
```

Note the unexpanded `${jmri.log.path}` in the destination.

PUlvestad, 2024-01-21:
My files show messages.log.1 (and messages.log.2) while his error shows messages.1.log

I don't see his error here

KenC57, 2024-01-21:
Well that permission error makes sense. It's trying to copy from the user
space to the program space. It's got a messed up path that's for sure.

dsand47, 2024-01-21:
The rolled messages file name was changed from messages.log.n to messages.n.log as part of the SLF4J project at 5.5.3.

dsand47, 2024-01-21:
Vinny provided the JMRI Context information:

jmri.log.path: C:\Users\User\JMRI\log\

PUlvestad, 2024-03-11:
Issue #12955 is probably related

PUlvestad, 2024-03-15:
There is another reported issue with message logs growing past 1MB
https://groups.io/g/jmriusers/message/228232

bobjacobsen, 2024-08-26:
I believe this was fixed in JMRI 5.8.  Could you try that release please?

PUlvestad, 2024-08-26:
It appears okay in 5.9.2plus

PUlvestad, 2024-08-26:
Fixed

closed 2024-08-26 by PUlvestad