JMRI/JMRI #12864: Programming Error

RLDvorak, 2024-01-27:
When running JMRI DecoderPro with my WIndows computer connected to my Digitrax DCS52 controller I get the following error when trying to read the CV settings of a loco on the programming track:
"programming error: No acknowledgement of locomotive (308)"
I know I have good connection between computer and controller, I just updated the firmware on the DCS52 successfully.

danielb987, 2024-01-28:
Have you tried another loco? Ensured that there is good contact between loco and rail? If there is bad contact between decoder and rail, the loco might still be able to run but not being able to read CV values.

danielb987, 2024-01-29:
@RLDvorak 
Is this thread on the JMRI users list about the same error as this issue is about?
https://groups.io/g/jmriusers/topic/104019439

closed 2024-04-22 by github-actions[bot]