JMRI/JMRI #12865: Turnout table marks sensor feedback fields with red border

dsand47, 2024-01-29:
This occurs when the table is sorted descending by system name or either direction for user names.

See discussion at https://groups.io/g/jmriusers/message/226147.

Git Bisect results:

```
das@mba15 JMRI % git bisect good
b66a54f09e6c749257657c40c5a4ff7a8a38e930 is the first bad commit
commit b66a54f09e6c749257657c40c5a4ff7a8a38e930
Author: Steve Young <icklesteve@yahoo.com>
Date:   Thu Jun 29 16:42:02 2023 +0100

    Update TurnoutTableJTable.java
    
    Sensor 1 and Sensor 2 columns display a red border when incorrectly selected

 .../beantable/turnout/TurnoutTableJTable.java      | 29 ++++++++++++++++++++++
 1 file changed, 29 insertions(+)
```

bobjacobsen, 2024-01-29:
This might be a case of needing to use [JTable.convertRowIndexToModel( int viewIndex )](http://java.sun.com/javase/6/docs/api/javax/swing/JTable.html#convertRowIndexToModel%28int%29)

bobjacobsen, 2024-01-29:
This might be an example of needing to use [JTable.convertRowIndexToModel( int viewIndex )](http://java.sun.com/javase/6/docs/api/javax/swing/JTable.html#convertRowIndexToModel%28int%29)