JMRI/JMRI #12892: Operations Pro: enhancement suggestion: ability to restrict tracks by division

tomgmyrick, 2024-02-07:
Currently when editing a track, you are able to restrict/allow which cars are allowed on the track by "Road Option" and "Load Option".  My suggestion is to also add "Division Option" so that cars can be restricted/allowed by their division.

I have attached a screenshot of my RIP track for my CP yard (CP division).  I only want cars that belong to the CP division to be allowed on the RIP track.  In order to do this, I needed to determine what roads belonged to the CP division, and restrict by the roads.  This worked in this case because I don't have any road names for cars that cross divisions, but it could happen (for example TTX cars could belong to either my CP or CN divisions in the future). 

I would go further and add the division option anywhere you can restrict/allow by road or load. 
[image]

DanielBoudreau, 2024-02-07:
Hi Tom,
At first glance your request seem quite reasonable.  However after some thought there are issues.  When it comes to divisions, it controls car movement, but more specifically empty cars.  A loaded car can travel to any division, independent of the car's division.  Empty cars can only return to its division yard, spur or staging.  So if you have a RIP track (spur) at a division location, empty cars for that division are only allowed.  Empty cars from other divisions will not use that RIP track.  Loaded cars from any division could use the RIP track.  Now the issues. If the GUI allowed you to specify which divisions are allowed to a track, things get complicated.  Say you had a location assigned to division "East".  A RIP track at division East would accept empty cars that were division East, and loaded cars from any division.  Now say you told the program to accept cars from division "West".  Now the RIP track would only accept loaded cars from division West, and no empty cars from any division. If we allow more divisions to be added, say "North" loaded cars from division North would be allowed.  But empty cars from East, West and North are never accepted.  So the divisional control is only for loaded cars?

tomgmyrick, 2024-02-07:
Hi Dan,

You make excellent points.  This is not so straightforward.

I was just reading up on the prototype and a RIP track can have cars from any division or railroad.  Basically, if a car travels on the railroad's line and needs minor repairs, they are done on the RIP, regardless of the owner of the car.   The way JMRI currently works allows for 2 scenarios in my situation:

  1.  No restrictions on cars that can go to the RIP.  Loaded and empty cars from the home division can go to the RIP, but only loaded cars from a foreign division can use the RIP.  Empty cars from the foreign division are not allowed.

  1.  Restrictions by road are placed on the RIP track.  The roads used are associated with the home division.  This means that loaded and empty cars from the home division can use the RIP, and no cars from a foreign division can use it.

Neither scenario is prototypical.  The RIP track should be able to service loaded and empty cars, regardless of division.  If a car needs repairs, it is repaired at the closest facility.

My request to be able to restrict by division would not be prototypical either for the reasons you mentioned.  I was providing a solution to a problem I did not see (users do this all of the time 🙂).

The real problem is how can a track be set up to allow loaded and empty cars from any division?  This is what a RIP track needs to do, if I am understanding correctly.

My current setup restricting by road is probably the closest I can get to the prototype currently.  The reason is that both the CN and CP yards are in the same city, so I don't think it is a stretch to expect a car that needs to be repaired to be sent down the track a bit to their home yard in this case.

Is my understanding correct?

Regards,

Tom Myrick

www.tominator.ca<http://www.tominator.ca/>

tomgmyrick, 2024-02-08:
Hi Dan,

That's a good point about just getting loaded cars from the other division.  I actually want very little traffic for the RIP track.  It is in a track pool with a yard track, and I want the yard track to get the majority of the work.  I probably just want to limit the RIP track to one car occasionally.  I can work with that.

However, there still may be a case for possibly another type of track called "RIP".  RIP tracks are different than spurs or yards.  There are workarounds but they are workarounds.  I am not sure how many people use RIP tracks in their layouts, so not sure if it's worth the effort or not.

Regards,

Tom Myrick

www.tominator.ca<http://www.tominator.ca/>

Hi Tom,

There’s a workaround for number 1.

You create another “same” location for your RIP track but for the other division. Now empty cars for the other division can use the RIP track. You need to split some of the RIP length into each location. You place the extra location in your train’s route. This solves the only empty cars for a division are allowed.

You could of course ignore the fact that empty cars from the other division are never serviced by the RIP track. Since you getting loaded cars, there is traffic from the other division. Not sure how long your RIP track is or how much traffic you really want.

Dan

DanielBoudreau, 2024-02-08:
Tom,

I’d like to close out this request for an enhancement.  I understand the possible need for a special RIP track, not sure worth my time coding.

Dan

tomgmyrick, 2024-02-08:
Hi Dan,

Agreed.

Regards,

Tom Myrick

www.tominator.ca<http://www.tominator.ca/>

closed 2024-02-08 by DanielBoudreau