JMRI/JMRI #12899: Confusing "Scale" dialog in Panel Editor

bobjacobsen, 2024-02-12:
TLDR: The way the "scale" dialog window works is confusing.

This is the dialog that's reached from the "scale" item in the contextual menu on a Panel Editor icon:

[image]

It always comes up showing "100", even if the icon has been previously scaled and a scaling value stored in the panel file, In this example, the icon is already at 125% and is displaying on the panel that way:
```
<turnouticon turnout="ZE P7 request" x="276" y="126" level="7" forcecontroloff="false" hidden="no" positionable="true" showtooltip="true" editable="true" tristate="false" momentary="true" directControl="false" class="jmri.jmrit.display.configurexml.TurnoutIconXml">
      <icons>
        <closed url="program:resources/icons/USS/sensor/red-on.gif" degrees="0" scale="1.25">
          <rotation>0</rotation>
        </closed>
        <thrown url="program:resources/icons/USS/sensor/green-on.gif" degrees="0" scale="1.25">
          <rotation>0</rotation>
        </thrown>
        <unknown url="program:resources/icons/USS/sensor/s-unknown.gif" degrees="0" scale="1.25">
          <rotation>0</rotation>
        </unknown>
        <inconsistent url="program:resources/icons/USS/sensor/s-inconsistent.gif" degrees="0" scale="1.25">
          <rotation>0</rotation>
        </inconsistent>
      </icons>
      <iconmaps />
    </turnouticon>
```

But the dialog is opening at 100%.

The operation is consistent with a "tell me how much you want to scale the _existing_ size icon", e.g. a relative sizing based on the existing 125%.  But then that "100" shown to center left is just confusing....

mstevetodd, 2024-02-13:
@bobjacobsen, I agree this is very confusing.

I've been trying to help an AnyRail user with the export to JMRI, and ran into this as well. AnyRail sets "scale" for the exported background image to a calculated decimal value based on the AnyRail drawing width and the current monitor resolution. So scale (in the xml) may be "0.2832" or somesuch. But I could never find that value in the UI, because of this issue.

(Warning, maybe off-topic for this issue) Adjusting the scale from 100% seems to start with the already-scaled image in memory rather than going back to the image on the file system. So, initially displaying the image at scale 0.2832 (understandably) squishes the image to unreadability, but subsequently changing the scale from the UI (say to 200%) just expands the unreadable blotches.

I think we should show the scale value itself, since that is what we store. 
And I also think we should go back to the file system image when scaling to get the best results.

cprail1, 2024-02-19:
So trying to use this in a consistent way, I ask myself is the word “scale” in that dialog box a verb or a noun?

I am expecting it to be a noun:  “The current scale is … and the new one should be …”

However perhaps the designer meant it a verb: “How much should we scale the current appearance by now?”  So it starts at 100, you change that to 200 to double the size it’s at _now_, not the original size. 

Although I consider the noun to be the more intuitively obvious and thus simpler, trying to use this as a verb doesn't seem to work logically either - entering 100 in the box, ie exactly the same figure as displayed when one opens the window and in the same box, when the xml says the scale is at 1.5, should leave the xml at 1.5 by the verb logic. 

However, if you enter 100, it reverts to a scaling of "1.0" and thus reduces the item even though you've not requested this. Entering what is already there shouldn't make a change in any common sense gui. So I'm not sure it is working as intended in either interpretation.  This is where I decided it was a bug in either usage. 

Personally, I'd prefer this to simply be a correct reflection, in %, of what the xml itself shows in the scale i.e
" <closed url="program:resources/icons/USS/sensor/red-on.gif" degrees="0" scale="1.25">" 
...should present 125% and changing it to 150%, say, should change the scale="1.25" to scale="1.5"

Anyway, I hope this can be changed to work logically in one interpretation or the other. Currently it is not following either interpretation consistently, IMHO.

danielb987, 2024-10-21:
What's the status of this issue?

bobjacobsen, 2024-10-21:
I have an improvement (not really a fix) for this being tested, but I haven't heard back.  Will ping.

bobjacobsen, 2024-10-21:
My modification didn't work.  The `setSelectionsScale` method in `jmri.jmrit.display.Editor` is used both for overall scaling and for scaling a group of selected elements.  In that second case, in some cases there isn't a consistent value for the current scaling:  it can differ from element to element.  It's going to take a bit of redesign to figure out how to handle this.

bobjacobsen, 2024-10-24:
After some additional study, I don't think there's a meaningful way to handle the "Scale Percentage" (i.e. current value) display.  Sometimes it can have multiple values.  

I propose to remove that, and have the scaling operation act as a factor on the existing scaling of all the selected items.  E.g. if you put 200 there, all the selected items will have their scaling increased by a factor of two:  100% -> 200%, 150% -> 300%, 40% -> 80%. 

This is the "verb" version of the dialog, which makes the most sense to me.

Comments welcome.  I'll probably implement this sometime this weekend.

cprail1, 2024-10-24:
Bob, Your proposed logic is how I intuitively expected it to work when I came across this originally so, personally, I think that will be the easiest for a typical user to understand and expect as a behavior.  Cheers, Peter

closed 2025-02-12 by silverailscolo