JMRI/JMRI #12906: LogixNG Listen on Beans combined with Sensor Groups can result in null sensors in LogixNG

dsand47, 2024-02-14:
A long but descriptive title.

The null sensor occurs randomly.  

The attached zip file has the layout data xml file and a TSV file.

When clicking on a number, there is a five second delay to simulate the feedback.  Wait for the platform line to change color before selecting another number.

[TransferTable.zip](https://github.com/JMRI/JMRI/files/14285158/TransferTable.zip)

danielb987, 2024-02-16:
I presses the numbers and wait for the platform line to change color. Then I do a new try and then again, ...
But I'm unable to get the error.

Could you try to track down the error? I assume the error occurs in the LogixNG action `Listen on beans`. If so, where in this class? On which line? A stacktrace might help.

dsand47, 2024-02-17:
> I presses the numbers and wait for the platform line to change color. Then I do a new try and then again, ... But I'm unable to get the error.
>
> Could you try to track down the error? I assume the [...]

I don't get an error since I check for a null result.  You will see the messages in the system console.

danielb987, 2024-02-17:
> I don't get an error since I check for a null result. You will see the messages in the system console.

Is it this message you mean?
`00:35:02,248 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-3 [JMRI LogixNGThread]`

I still don't understand what I should be looking for.

dsand47, 2024-02-18:
> > I don't get an error since I check for a null result. You will see the messages in the system console.
> 
> Is it this message you mean? `00:35:02,248 jmri.jmrit.logixng.actions.LogData WARN - -- selection is S-Select-3 [JMRI LogixNGThread]`
> 
> I still don't understand what I should be looking for.

Eventually you will get a null response after the valid response.
```
18:25:19,254 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-2 [JMRI LogixNGThread]
18:25:26,451 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-4 [JMRI LogixNGThread]
18:25:32,221 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-1 [JMRI LogixNGThread]
18:25:38,198 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-5 [JMRI LogixNGThread]
18:25:38,207 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is null [JMRI LogixNGThread]
```

In this case, sensor 5 was the expected sensor, followed by the null value 9 ms later.

When an inactive group sensor is clicked , it goes active and previously active sensor goes inactive.  I find it interesting that we don't see the sensor that is going inactive.  If the active sensor is set inactive, then it will show up.

danielb987, 2024-02-18:
Could you try my branch `danielb987 : LogixNG_SensorGroups`?
https://github.com/danielb987/JMRI/tree/LogixNG_SensorGroups

Or add the two changes manually to your branch? It's only two logging statements I have added.
https://github.com/JMRI/JMRI/compare/master...danielb987:JMRI:LogixNG_SensorGroups

It will show what's happening. What events do `ActionListenOnBeansTable` sees and what values does the local variables have when it's executed.

The question is if something goes wrong before `ActionListenOnBeansTable` sees the property change event or if something goes wrong in LogixNG.

dsand47, 2024-02-18:
> Could you try my branch `danielb987 : LogixNG_SensorGroups`? https://github.com/danielb987/JMRI/tree/LogixNG_SensorGroups
> 
> Or add the two changes manually to your branch? It's only two logging statements I have added. [master...danielb987:JMRI:LogixNG_SensorGroups](https://github.com/JMRI/JMRI/compare/master...danielb987:JMRI:LogixNG_SensorGroups)
> 
> It will show what's happening. What events do `ActionListenOnBeansTable` sees and what values does the local variables have when it's executed.
> 
> The question is if something goes wrong before `ActionListenOnBeansTable` sees the property change event or if something goes wrong in LogixNG.

Here is the log info.  I added a "this" so that you would know which listen on beans created the message.

```
     [java] 15:21:59,739 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: propertyChange: Prop name: KnownState, source: IS101, old: 2, new: 4 [AWT-EventQueue-0]
     [java] 15:21:59,740 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: propertyChange: Prop name: KnownState, source: IS104, old: 4, new: 2 [AWT-EventQueue-0]
     [java] 15:21:59,740 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: Execute: Prop name: KnownState, source: S-Select-1, new: 4 [JMRI LogixNGThread]
     [java] 15:21:59,740 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-1 [JMRI LogixNGThread]
     [java] 15:21:59,746 jmri.jmrit.logixng.actions.LogData    WARN  - ---- Selection S-Select-1 is not active [JMRI LogixNGThread]
     [java] 15:21:59,747 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: Execute: Prop name: KnownState, source: S-Select-4, new: 2 [JMRI LogixNGThread]
     [java] 15:21:59,747 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-4 [JMRI LogixNGThread]
     [java] 15:21:59,760 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: propertyChange: Prop name: KnownState, source: IT104, old: 2, new: 4 [AWT-EventQueue-0]
     [java] 15:21:59,760 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: Execute: Prop name: KnownState, source: T-Track-4, new: 4 [JMRI LogixNGThread]
     [java] 15:22:04,771 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0044 :: propertyChange: Prop name: KnownState, source: IS301, old: 2, new: 4 [AWT-EventQueue-0]
     [java] 15:22:04,772 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0044 :: propertyChange: Prop name: KnownState, source: IS304, old: 4, new: 2 [AWT-EventQueue-0]
     [java] 15:22:04,788 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: propertyChange: Prop name: KnownState, source: IT104, old: 4, new: 2 [AWT-EventQueue-0]
     [java] 15:22:04,789 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0044 :: Execute: Prop name: KnownState, source: S-Feedback-4, new: 2 [JMRI LogixNGThread]
     [java] 15:22:04,820 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: Execute: Prop name: KnownState, source: T-Track-4, new: 2 [JMRI LogixNGThread]

     [java] 15:22:09,323 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: propertyChange: Prop name: KnownState, source: IS104, old: 2, new: 4 [AWT-EventQueue-0]
     [java] 15:22:09,323 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: propertyChange: Prop name: KnownState, source: IS101, old: 4, new: 2 [AWT-EventQueue-0]
     [java] 15:22:09,323 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: Execute: Prop name: KnownState, source: S-Select-1, new: 2 [JMRI LogixNGThread]
     [java] 15:22:09,323 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-1 [JMRI LogixNGThread]
     [java] 15:22:09,339 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: propertyChange: Prop name: KnownState, source: IT101, old: 2, new: 4 [AWT-EventQueue-0]
     [java] 15:22:09,339 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: Execute: Prop name: null, source: null, new: null [JMRI LogixNGThread]
     [java] 15:22:09,339 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is null [JMRI LogixNGThread]
     [java] 15:22:09,344 jmri.jmrit.logixng.actions.LogData    WARN  - ++ selection was null [JMRI LogixNGThread]
     [java] 15:22:09,344 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: Execute: Prop name: KnownState, source: T-Track-1, new: 4 [JMRI LogixNGThread]
     [java] 15:22:14,360 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0044 :: propertyChange: Prop name: KnownState, source: IS304, old: 2, new: 4 [AWT-EventQueue-0]
     [java] 15:22:14,360 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0044 :: propertyChange: Prop name: KnownState, source: IS301, old: 4, new: 2 [AWT-EventQueue-0]
     [java] 15:22:14,364 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: propertyChange: Prop name: KnownState, source: IT101, old: 4, new: 2 [AWT-EventQueue-0]
     [java] 15:22:14,365 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0044 :: Execute: Prop name: KnownState, source: S-Feedback-1, new: 2 [JMRI LogixNGThread]
     [java] 15:22:14,401 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: Execute: Prop name: KnownState, source: T-Track-1, new: 2 [JMRI LogixNGThread]
```

danielb987, 2024-02-18:
Note to myself:
```
1: 15:22:09,323 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: propertyChange: Prop name: KnownState, source: IS104, old: 2, new: 4 [AWT-EventQueue-0]
2: 15:22:09,323 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: propertyChange: Prop name: KnownState, source: IS101, old: 4, new: 2 [AWT-EventQueue-0]
3: 15:22:09,323 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: Execute: Prop name: KnownState, source: S-Select-1, new: 2 [JMRI LogixNGThread]
4: 15:22:09,323 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is S-Select-1 [JMRI LogixNGThread]
5: 15:22:09,339 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0061 :: propertyChange: Prop name: KnownState, source: IT101, old: 2, new: 4 [AWT-EventQueue-0]
6: 15:22:09,339 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: Execute: Prop name: null, source: null, new: null [JMRI LogixNGThread]
7: 15:22:09,339 jmri.jmrit.logixng.actions.LogData    WARN  - -- selection is null [JMRI LogixNGThread]
8: 15:22:09,344 jmri.jmrit.logixng.actions.LogData    WARN  - ++ selection was null [JMRI LogixNGThread]
```
At line 1, you get a propertyChange event.
At line 2, you get a new propertyChange event for the same action.
At line 3, the ConditionalNG executes.
At line 6, the same ConditionalNG executes again, without a new propertyChange event. Therefore everything is null.

It seems that two events happens at the same time, before the ConditionalNG executes, and that that's the cause of the problem.

dsand47, 2024-02-18:
> Note to myself:
>
> ```
> 1: 15:22:09,323 ixng.actions.ActionListenOnBeansTable WARN  - IQDA:AUTO:0008 :: propertyChange: Prop name: KnownState, source: IS104, old: 2, new: 4 [AWT-EventQueue-0]
> 2: [...]

Lines 1 and 2 are different beans.  The first one is going inactive and second is going active.  Normally there will be an execute for each one in the same sequence.  This is shown in the first set of messages.

In the second set, the events are out of order.  The second bean executes before the first bean.  The first bean reference is null when it finally executes.

danielb987, 2024-02-18:
> Lines 1 and 2 are different beans.

Yes, different sensors. But I ment they are the same LogixNG action IQDA:AUTO:0008. The problem is that the action doesn't buffer propertyChange events. If two events arrive before the action executes, this might happen.

dsand47, 2024-02-18:
While the possibility of two events at the same time is very small it is not zero.  A sensor group is probably the most likely to cause this problem.

danielb987, 2024-02-23:
> While the possibility of two events at the same time is very small it is not zero. A sensor group is probably the most likely to cause this problem.

I'm working on this but the problem is not so easy to fix.

danielb987, 2024-02-25:
@dsand47 
PR  #12925 fixes this bug.

It's difficult code to get right and I don't have much tests for these classes so it would be good if you could test all three: ActionListenOnBeans, ActionListenOnBeansLocalVariable and ActionListenOnBeansTable.

Thanks!

dsand47, 2024-02-27:
> @dsand47 PR #12925 fixes this bug.
> 
> It's difficult code to get right and I don't have much tests for these classes so it would be good if you could test all three: ActionListenOnBeans, ActionListenOnBeansLocalVariable and ActionListenOnBeansTable.
> 
> Thanks!

ActionListenOnBeans and ActionListenOnBeansTable are okay.  

ActionListenOnBeansLocalVariable has a problem.  The first selection works fine.  The second selection causes an exception when the first one goes inactive.

```
     [java] 22:38:17,430 jmri.jmrit.logixng.actions.LogData    WARN  - New state for S-Feedback-2 is 2 [JMRI LogixNGThread]
     [java] 22:38:22,859 jmri.jmrit.logixng.actions.LogData    WARN  - New state for S-Feedback-2 is 4 [JMRI LogixNGThread]
     [java] 22:38:22,860 g.implementation.DefaultConditionalNG WARN  - ConditionalNG IQC:AUTO:0008 got an exception during execute: java.lang.NullPointerException [JMRI LogixNGThread]
     [java] java.lang.NullPointerException: null
     [java] 	at jmri.jmrit.logixng.actions.ActionListenOnBeansLocalVariable$InternalFemaleSocket.execute(ActionListenOnBeansLocalVariable.java:387) ~[classes/:?]
     [java] 	at jmri.jmrit.logixng.implementation.DefaultConditionalNG.internalExecute(DefaultConditionalNG.java:173) ~[classes/:?]
     [java] 	at jmri.jmrit.logixng.implementation.DefaultConditionalNG.lambda$execute$0(DefaultConditionalNG.java:145) ~[classes/:?]
     [java] 	at jmri.jmrit.logixng.util.LogixNG_Thread.lambda$new$0(LogixNG_Thread.java:148) ~[classes/:?]
     [java] 	at java.lang.Thread.run(Thread.java:829) [?:?]
  ```

Here is the test xml file.

[ActionListenOnBeansLocalVariable.txt](https://github.com/JMRI/JMRI/files/14414452/ActionListenOnBeansLocalVariable.txt)

closed 2024-02-27 by danielb987