JMRI/JMRI #12911: Request Update On All Sensors from LOGIXNG  doesn't work for CMRI

slolitzky, 2024-02-18:
I manually change the status of my sensor icons on my panel for debugging logixng.  To get them back to the hardware state I have simple logixng that when a sensor on my panel goes active makes the 

REQUEST UPDATE ON ALL SENSORS

I select to request only on CMRI sensors.

The JMRI sensor states are not updated to match the hardware.  If I  use a script with the call to 

requestUpdateFromLayout() 

This works perfectly.  I know that there had been some problems with the query button on the sensor table for CMRI.  See Issue #12305 and #12306.

Using PanelPro 5.5.6+Re5414c68FF

danielb987, 2024-02-18:
> The JMRI sensor states are not updated to match the hardware. If I use a script with the call to
> requestUpdateFromLayout()
> This works perfectly.

Do you do requestUpdateFromLayout() on a particular sensor? If so, there is a LogixNG action that requests update of a single sensor. Not all connections support requesting all sensors.

slolitzky, 2024-02-19:
In my script I iterate thru all of the sensors and use

            mysen.requestUpdateFromLayout()

In logixng I used

    Request update off all sensors // doesn't appear to work

but I don't use

    Request update of sensor
---------------
I'm ok using a script to clear all of my sensors, if this is not a bug just
not implemented for CMRI

icklesteve, 2024-02-21:
The loop all Sensors, call DigitalIO method requestUpdateFromLayout() from each is pretty much the CbusSensorManager implementation of  SensorManager.updateAll(), which has worked for a good while with no issues.

I'm wondering if this should be the default AbstractSensorManager updateAll() to enable hardware types that have implemented the single Sensor requestUpdateFromLayout() to query via updateAll() ?

I can't see any cons to this at present?
If I don't see any posted in the next few days I'm minded to see what a PR with this change looks like.

Steve.

bobjacobsen, 2024-02-21:
On some systems, like OpenLCB/LCC, this will generate a lot of traffic on the line.  But I don't see that as a reason to not do this:  if people don't like that traffic, they can avoid doing the operation.  So I think this is a good idea.

KenC57, 2024-02-22:
Daniel,

Since the CMRI system is constantly polling all the nodes, I don't see
the idea of requesting an update would apply. In well under a second
it would have received a reply from all the nodes anyway.

bobjacobsen, 2024-02-22:
The CMRI connection caches the value so that you can override it locally. E g if it’s ACTIVE and you click to set it INACTIVE, it’ll stay that way until either the remote changes ACTIVE to INACTIVE to ACTIVE again or you click again. Very similar to how other systems behave.

icklesteve, 2024-02-24:
Hi @slolitzky , 

In #12916 the main SensorManager updateAll method was tweaked so this should now work as expected for CMRI.

If you fancied testing this before the next JMRI test release, build 1786 onwards has the change at 
https://builds.jmri.org/jenkins/job/Development/job/packages/

Please let us know how you get on,
Steve.

slolitzky, 2024-02-29:
Sorry it took so long to try out the fix.
The fix works.  I have noticed that it takes noticeably longer to reread
the sensors and clear the block status on my panel than when I used the
script. You can see the blocks visually clearing one at a time.

My script runs and all of the sensors clear at the same time.  There is no
lag in the update of the panel display.  All the block statuses change at
the same time.

Here is the script code:

import jmri
for sen in sensors.getNamedBeanSet():
     print "Sensor: " + str(sen.systemName)
     print sen.requestUpdateFromLayout()

I appreciate you fixing this problem but I will use my script for now
because of the visual delay.

Seth Olitzky

icklesteve, 2024-03-05:
Hi @slolitzky ,

The delay between sending Sensor requests avoids overloading certain connection types.
The delay length can be set in Edit > Preferences > Connections, Connection > Additional Connection Settings.
Sometimes the default delay can be quite high, 50ms works well for me on a CAN Network, YMMV.

Closing issue as OP advises fix works.

closed 2024-03-05 by icklesteve