JMRI/JMRI #12932: popup menu not working in control panel editor

jfalkenb, 2024-02-27:
I was setting up a new layout in 5.6 release, and noticed that the popup menu (right-click) is not working.

I checked earlier versions, and noticed that the problem occurred at the change from 5.0 to 5.2.

What can be done?

Thanks in advance

Hans

bobjacobsen, 2024-02-27:
Exactly how are you trying to access it?  With what type of computer?

With JMRI 5.7.4 on a Mac, I get the pop-up menu with mouse-down while holding "option".

dsand47, 2024-02-27:
Hans,

Make sure the panel is in edit mode.  The popup menu is disabled for most items when not in edit mode.

Dave Sand

jfalkenb, 2024-02-29:
Dave,

 

 

I checked a bit more. 

On my pc (windows 10) I now also installed release 5.4. Everything is ok with respect to editing.

I noticed during this step by step upgrades that I lost some logixNG. And in the logix some logix were disabled suddenly. Not a big issue for now, but in the future??? Any hints from your side?

 

The editing on my Rpi 4 is still a problem. (both are 4 running on bullseye, not the standard new one (issues with installing the correct java))

Attached a screenshot.

This RPi4 is connected to a simple layout for some time (editing was no issue). Maybe the change from RPi3 could be the moment the problems started. I did not check at that time.

 

The new RPi4 has the same problem. Most likely from the start? Not sure.

 

 

I think I am all the time in editing mode. Or am I incorrect?

 

 

Thanks for your advice,

 

 

Hans

dsand47, 2024-02-29:
Hans,

For Control Panel Editor, look at the File menu.  If the only menu item is "Open Editor" then you are not in edit mode.

I don't see a screen shot.

You can install Java 11 on Bookworm.  See https://groups.io/g/jmriusers/message/222531

The layout data xml files are not always backwards compatible.  If you are not watching the system console, you might miss some errors.

Dave Sand

jfalkenb, 2024-03-01:
Dave

Sorry, I used my email, maybe that is the reason the file dropped. I try via the web now.

[image]

I know backwards incompatibility. That's why I used old files with the correct release of jmri. For some time I added the release version in the file name (easier to see the release I worked on).

Thanks

Hans

dsand47, 2024-03-01:
Hans,

If you change the "view" from control panel editor to panel editor does the context menu work?

Dave Sand

JMRI.Screenshot.2024-02-29.110549.png (view on web) <https://github.com/JMRI/JMRI/assets/50291241/32267e8b-742c-48cb-98c8-42099ca6d737>

dsand47, 2024-03-01:
The problem is caused by a change in JMRI 5.1.2.

```
das@mba15 JMRI % git bisect bad    
3f5855ebf6196d385d0f70cd146b7a55a6499dbc is the first bad commit
commit 3f5855ebf6196d385d0f70cd146b7a55a6499dbc
Author: Daniel <db123@bergqvist.se>
Date:   Tue Jul 5 00:47:08 2022 +0200

    Update JmriMouseEvent.java

 java/src/jmri/util/swing/JmriMouseEvent.java | 18 ++++++++++++------
 1 file changed, 12 insertions(+), 6 deletions(-)
```

The change only affects Linux and Control Panel Editor.  Changing the panel view to Panel Editor is a work around for the problem.

jfalkenb, 2024-03-02:
I tried again. 
And surprised.

When in Control Panel Editor: Sensors and turnouts can be triggered upon mouse click. Editing of icons is not possible.

Change view to Panel Editor (this window with "Rename Panel:" etc, has to be opened all time. Don't close it first)
Some observations.:
Editing of icons is now possible.
If sensors are clicked, nothing happens. If the sensor is dragged a little bit, it changes the state.
Turnout click:: apparently a signal is given to the layout (can hear the relais), but double signal and state stays the same. dragging TO works, like the sensor.

Closing the Control Panel editor via the file menu, will create a new panel. The old (original) panel will be blank. It looks like that when the file is saved and loaded after restart, the file looks fine (not extensively tested).
Panel  stays in edit mode, but again not possible to edit.

Closing the Control Panel editor via the close window (top right of the window). The Control panel goes to non edit mode. No additional panel created. But still can edit.
Can still not operate sensors and TO.

Looks like that after saving the file and reloading, the sensors and TO still cannot be activated as per normal. First you have to close the "Change view to Control Panel Editor" via the file menu. Than sensors and turnouts work. But an additional panel is created.

Confusing

In release 5.0 this was not an issue
Any suggestions?

jfalkenb, 2024-03-14:
no more suggestions?

dsand47, 2024-04-16:
It has now been confirmed that this change also affects automatically creating connecting Layout Editor track segments between turnout end points.  This affects Windows and Linux.

dsand47, 2024-04-17:
The attached spreadsheet documents the "real" mouse events for LE, PE and CPE panels using macOS, Linux and Windows.

JMRI has overrides for two of the mouse states, `isPopupTrigger()` and `isMetaDown()`.  These are causing the problems identified here.  

To fix the problems will require changes to the overrides and/or changes to the editor code, without breaking anything that currently works.

[jmri-mouse-buttons.xlsx](https://github.com/JMRI/JMRI/files/15017830/jmri-mouse-buttons.xlsx)

closed 2024-04-19 by danielb987