JMRI/JMRI #12950: DCC-EX connection sending delay should be interrupted when incoming message seen

mstevetodd, 2024-03-08:
The DCC-EX transmitLoop() currently has an initial 1.5s delay. This was added because the Arduino will reboot if it receives data before it is ready to receive. Since JMRI will retry messages for a while, this resulted in a reboot loop until JMRI timed out of retries. The 1.5s delay "fixed" that for a while.
Recently, a user reported (https://groups.io/g/jmriusers/message/227777) what turned out to be this reboot loop with one computer (but not another). Increasing the time (as a test) to 10s stopped the behavior. A check of the debug log showed that Arduino took 9s to start sending data (vs. < 1s for another computer).

1) So, I suggest we make the initial sending delay a configuration setting, with the default at 1500ms.

2) Also, I suggest adding a way to interrupt the delay if an incoming message is received. The idea being that if the Arduino is able to send, it should be ready to receive. If this is implemented, perhaps the default in 1) could be increased to 10000ms.

mstevetodd, 2024-03-08:
This is the sleep I want to be able to change, and to interrupt: https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/dccpp/DCCppTrafficController.java#L30

bobjacobsen, 2024-03-09:
I can work on adding a configuration option if nobody else wants to do that. But do you even need that if you can listen for the first message being received.  Just set that delay to Unreasonably Long and let the first message end it.

One way to end the delay when a message is received is to break the delay up into a loop of e.g. 100 msec steps, checking some status boolean after each step for whether to continue on or loop again.  Have the receive thread set that boolean when the first message  has been received,  Do you know a logical place to put that?

mstevetodd, 2024-03-09:
Thanks, Bob.
My concern with setting a really high delay is that earlier versions of the DCC-EX and DCC++ software do not send any unsolicited messages, so they'd never break out of the delay. 
But I'd be ok with simply changing the timeout to 10s, which should be acceptable in all cases.
Thanks for the suggestion of checking a flag to get out of a delay loop. That makes sense.

bobjacobsen, 2024-03-09:
The `bobjacobsen/DCCpp-custom-delay` branch will let you pick between 1.5 seconds and 10 seconds at startup.  Don't have a way to actually run this, but the debugger says that it's working.

mstevetodd, 2024-03-09:
Thanks, @bobjacobsen 
I will give this a try with my hardware and advise.

mstevetodd, 2024-03-13:
> Thanks, @bobjacobsen I will give this a try with my hardware and advise.

@bobjacobsen Your change worked fine, and I've incorporated it into my PR with the change to interrupt the startup delay when possible. Thank you!

commit 2597e08b (2024-03-14): DCC-EX: interrupt send delay on incoming message, closes #12950
commit 70e7448d (2024-03-14): DCC-EX: interrupt send delay on incoming message, closes #12950
commit 77a93234 (2024-03-14): Merge pull request #12957 from mstevetodd/master

DCC-EX: interrupt send delay on incoming message, closes #12950
closed 2024-03-14 by danielb987