JMRI/JMRI #12955: StatusConsoleListener ERROR

nschertz, 2024-03-09:
Issue Type: __Bug__

During JMRI start, I've been getting the following error in the system console (line 2 below).  It doesn't seem to effect anything, but I'd like to remedy the program complaint.

ERROR StatusConsoleListener Unable to copy file C:\Users\nsche\JMRI\log\messages.log to C:\Program Files (x86)\JMRI\${jmri.log.path}messages.1.log: java.nio.file.AccessDeniedException C:\Users\nsche\JMRI\log\messages.log -> C:\Program Files (x86)\JMRI\${jmri.log.path}messages.1.log

I checked and both the message.log and session.log files are being created in the JMRI userfile directory.  Not sure why JMRI is attempting to copy them to the program directory.

PanelPro 5.6+R89a87446cb
OpenJDK 64-Bit Server VM Azul Systems, Inc. 11.0.17+8-LTS mixed mode
Windows 11 10.0 amd64

[session.log](https://github.com/JMRI/JMRI/files/14547509/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/14547510/systemInfo.txt)
[profile.zip](https://github.com/JMRI/JMRI/files/14547511/profile.zip)

bobjacobsen, 2024-08-26:
I believe this was fixed in JMRI 5.8.  Could you please try that release?

nschertz, 2024-08-27:
Bob -
I'm currently on JMRI 5.8 and haven't seen this issue occur..so it appears to be fixed.
Norb

closed 2024-08-27 by bobjacobsen