JMRI/JMRI #12960: NPE when creating an LCC turnout

rm5248, 2024-03-15:
Trying to create a turnout for LCC fails.

Start PanelPro, go to Tools->Tables->Turnouts, press the 'Add' button and try to add a turnout with hardware address of 12.  The following stacktrace is printed in the log:

```
21:54:29,039 jmri.jmrix.openlcb.OlcbTurnout        ERROR - Can't parse OpenLCB Turnout system name: 12 [AWT-EventQueue-0]
21:54:29,042 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException: Cannot invoke "jmri.jmrix.openlcb.OlcbAddress.toEventID()" because "this.addrThrown" is null
	at jmri.jmrix.openlcb.OlcbTurnout.finishLoad(OlcbTurnout.java:107) ~[jmri.jar:?]
	at jmri.jmrix.openlcb.OlcbTurnoutManager.createNewTurnout(OlcbTurnoutManager.java:91) ~[jmri.jar:?]
	at jmri.managers.AbstractTurnoutManager.newTurnout(AbstractTurnoutManager.java:118) ~[jmri.jar:?]
	at jmri.managers.ProxyTurnoutManager.makeBean(ProxyTurnoutManager.java:55) ~[jmri.jar:?]
	at jmri.managers.ProxyTurnoutManager.makeBean(ProxyTurnoutManager.java:19) ~[jmri.jar:?]
	at jmri.managers.AbstractProvidingProxyManager.provideNamedBean(AbstractProvidingProxyManager.java:50) ~[jmri.jar:?]
	at jmri.managers.ProxyTurnoutManager.provideTurnout(ProxyTurnoutManager.java:64) ~[jmri.jar:?]
	at jmri.jmrit.beantable.TurnoutTableAction.createPressed(TurnoutTableAction.java:562) ~[jmri.jar:?]
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972) ~[?:?]
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313) ~[?:?]
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) ~[?:?]
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) ~[?:?]
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) ~[?:?]
	at java.awt.Component.processMouseEvent(Component.java:6626) ~[?:?]
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3389) ~[?:?]
	at java.awt.Component.processEvent(Component.java:6391) ~[?:?]
	at java.awt.Container.processEvent(Container.java:2266) ~[?:?]
	at java.awt.Component.dispatchEventImpl(Component.java:5001) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2324) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4833) ~[?:?]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) ~[?:?]
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575) ~[?:?]
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) ~[?:?]
	at java.awt.Container.dispatchEventImpl(Container.java:2310) ~[?:?]
	at java.awt.Window.dispatchEventImpl(Window.java:2780) ~[?:?]
	at java.awt.Component.dispatchEvent(Component.java:4833) ~[?:?]
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:775) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:720) ~[?:?]
	at java.awt.EventQueue$4.run(EventQueue.java:714) ~[?:?]
	at java.security.AccessController.doPrivileged(AccessController.java:399) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:747) ~[?:?]
	at java.awt.EventQueue$5.run(EventQueue.java:745) ~[?:?]
	at java.security.AccessController.doPrivileged(AccessController.java:399) ~[?:?]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) ~[?:?]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:744) ~[?:?]
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) [?:?]
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) [?:?]
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) [?:?]
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:90) [?:?]

```

Note that the GUI page does have some text that "LCC Turnouts use one of these patterns", but I've also tried putting in the complete LCC ID but it doesn't like that either, so I'm not sure exactly what it is looking for in terms of trunout ID.

Verified with JMRI 5.6.

dsand47, 2024-03-15:
The system name for a turnout consists of two event ids separated by a semi-colon.  The first one is the thrown event id and the second is closed event id.

11.22.33.44.55.66.77.88;11.22.33.44.55.66.77.99

The connection prefix and type will be added automatically resulting in MS11.22.33.44.55.66.77.88;11.22.33.44.55.66.77.99.  Be sure to provide a user name.

danielb987, 2024-03-15:
I think the error here is that `TurnoutTableAction.createPressed()` should show some sort of error message if the turnout couldn't be created.

`OlcbTurnoutManager.createNewTurnout()` could throw an `IllegalArgumentException` and the `TurnoutTableAction.createPressed()` could show a message dialog if an exception is thrown.

icklesteve, 2024-03-15:
If a System Name for a Managed Bean cannot be created, a BadSystemNameException should be thrown with the reason.
This reason is then displayed to the user, and the create button in the new Bean frame disabled.

OlcbTurnout expects a 2 part address, and throws the Exception if it receives a 1 part.
OlcbTurnoutManager returns a false positive when validating a 1 part address.

https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/openlcb/OlcbTurnoutManager.java#L152 is calling
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/openlcb/OlcbAddress.java#L320

In #12961 I've added a 2 part address validator to OlcbAddress for OlcbTurnoutManager and any other cases.

Steve.

rm5248, 2024-03-15:
> The system name for a turnout consists of two event ids separated by a semi-colon. The first one is the thrown event id and the second is closed event id. [...]

FWIW the reason I chose '12' was because I wanted to see if JMRI would translate the DCC accessory number into the correct LCC event ID, since there are well-known event IDs used for this purpose.  It would be convenient if this mapping of a normal DCC accessory address <-> LCC event ID existed.

bobjacobsen, 2024-03-15:
There's already a way to tell JMRI that you want to use the well-known turnout aliases:

From https://www.jmri.org/help/en/html/hardware/openlcb/index.shtml :
> As a special case, DCC turnouts connected via a bridge to an OpenLCB network can be addressed as e.g. MTT123. Note the 2nd T for "Turnout addressing". JMRI will then automatically generate the correct well-known events to throw/close the corresponding DCC turnout.

bobjacobsen, 2024-03-15:
> OlcbTurnout expects a 2 part address, and throws the Exception if it receives a 1 part.
> OlcbTurnoutManager returns a false positive when validating a 1 part address.

There's an exception to the "OlcbTurnout expects a 2 part address" rule, which is the e.g. MTT123 system name for DCC turnouts connected via a bridge.

icklesteve, 2024-03-15:
OlcbTurnout expects a 2 part address from OlcbAddress, see https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/openlcb/OlcbTurnout.java#L80

OlcbAddress converts, eg. MTT123 to a 2 part address which validates fine under #12961

rm5248, 2024-03-16:
> There's already a way to tell JMRI that you want to use the well-known turnout aliases:
>
> From https://www.jmri.org/help/en/html/hardware/openlcb/index.shtml :

Thanks!  In order to do this though, I had to go to Tools->Turnout Control and then put in MTT10 there, adding it from the 'Add' button on the turnout table doesn't work.

bobjacobsen, 2024-03-16:
> Thanks! In order to do this though, I had to go to Tools->Turnout Control and then put in MTT10 there, adding it from the 'Add' button on the turnout table doesn't work.

You put the "T123" in the _hardware_ name field; the `MT` (or whatever is the appropriate prefix) is added automatically.

PR #12961 (2024-03-16): OLCB Turnout Manager 2 part address validation
OlcbTurnoutManager - use 2 part address string validation
adds method validateSystemNameFormat2Part to OlcbAddress
fixes #12960
closed 2024-03-16 by danielb987