JMRI/JMRI #12962: Additional Information on Piko Decoders

bobjacobsen, 2024-03-16:
The attached files are additional information on the functioning of Piko decoders.  These were provided by Piko America with the help of Reed's Hobby Shop.

@uli-2016 You might find these useful.

[97442-90-7101_detailed instructions + CV chart 3-15-24.pdf](https://github.com/JMRI/JMRI/files/14625318/97442-90-7101_detailed.instructions.%2B.CV.chart.3-15-24.pdf)

[97440 ML4000 LightController CV Chart 3-15-24.pdf](https://github.com/JMRI/JMRI/files/14625319/97440.ML4000.LightController.CV.Chart.3-15-24.pdf)

[97442-90-7101_detailed instructions + CV chart.docx](https://github.com/JMRI/JMRI/files/14625320/97442-90-7101_detailed.instructions.%2B.CV.chart.docx)

[97440 ML4000 LightController CV Chart.docx](https://github.com/JMRI/JMRI/files/14625321/97440.ML4000.LightController.CV.Chart.docx)

uli-2016, 2024-03-28:
I will make the decoder file, but I dont make any tests.

uli-2016, 2024-04-02:
First pre-version was added with PR 13007. I can't do any tests with this decoder and this loco. Can anyone help?

ralflang, 2024-04-03:
I do have a Piko loco with a vendor default DCC decoder. Not sure if it matches the one you added though.

uli-2016, 2024-04-03:
Please read CV8 (= 162 ?) and CV7 (> 8 ?). In this case it is a Piko XP5.1 decoder, the way it should be.

closed 2024-12-02 by github-actions[bot]