JMRI/JMRI #12963: Script "\jython\LCC\NodeBackupToCSV.py" is missing an include statement

Cliff-Anderson, 2024-03-17:
As distributed, execution of this script fails
```
2024-03-17T15:42:33,066 jython.RunJythonScript                ERROR - Unable to execute script. [AWT-EventQueue-0]
javax.script.ScriptException: NameError: name 'JFileChooser' is not defined in <script> at line number 7
```

Suggested minimal fix is to insert two lines starting at line # 8 with:
```
import javax    ## 2024-03-17
from javax.swing import JFileChooser    ## 2024-03-17
```
which were slightly modified, but otherwise as directly copied from the companion script "\jython\LCC\CSVToNodeBackup.py"

Cliff-Anderson, 2024-03-17:
My ancient C programming habit bit me.
The title should have referred to missing **import statements**.

dsand47, 2024-03-18:
> @dsand47 Do you use LCC? The fix seems to be easy to do but I cannot test it since I don't use LCC.

The proposed change works fine.  The LCC backup text files are standard Java key/value property files.

closed 2024-03-19 by danielb987